Ultimaker/Cura #15749: Crash cura 5.3.1 sur windows 11 pro with extensions

stephanedegee, 2023-06-14:
### Application Version

5.3.1

### Operating System

Windows 11 PRO

### Printer

Biqu B1 SE PLUS

### Reproduction steps

Crash cura 5.3.1 sur windows 11 pro with extensions

### Actual results

Crash cura 5.3.1 sur windows 11 pro with extensions

### Expected results

no more crashes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Crash cura 5.3.1 sur windows 11 pro with extensions

MariMakes, 2023-07-17:
Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.

Hey @stephanedegee 

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

A Cura crash can have a lot of causes, so I'm not sure which one you are encountering. 
It would have helped if you filled in the form correctly.

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

stephanedegee, 2023-07-17:
Hello, 

 

Thank you for your answer. 

I found a solution by opening with a right click "as administrator" at this time I have access to the extensions menu without crashing. 

This is the only solution I have found.

 

Best regard

MariMakes, 2023-07-24:
Thanks for the update. 👍 

I'll add it to the list of proposed workarounds to help people get unstuck. 
I'll close this issue since you are not stuck anymore.