Ultimaker/Cura #15755: AppImage no longer works under Debian 12

kalsan, 2023-06-15:
### Application Version

Tested with 5.0.0 and 5.3.1

### Operating System

Debian 12 Bookworm

### Printer

irrelevant

### Reproduction steps

1. Download the AppImage
2. Attempt to run it

### Actual results

```
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open r600: /usr/lib/dri/r600_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: r600
libGL error: MESA-LOADER: failed to open r600: /usr/lib/dri/r600_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: r600
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_cura-lFS4h4n/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

Error appears claiming: "Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers."

The **functionning workaround** for this is to run the Appimage with the following environment variable set:

```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 
```

### Expected results

The AppImage should be able to run without the `LD_PRELOAD` env var set.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Note that this is not an issue related to OpenGL or missing drivers, as adding the environment variable solves the problem.

[image]

barbaraBrzozka, 2023-07-11:
I had the same problem, but switching to -modern AppImage release fixed it.

kalsan, 2023-07-12:
I can confirm Cura 5.4.0 "Linux Modern" fixes the problem for me as well.

closed 2023-07-12 by kalsan