Ultimaker/Cura #15756: Using 2. extruder on IDEX results in headcrash after printing raft

raspstar, 2023-06-15:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Sovol SV04 Dual

### Reproduction steps

Slicing with use of 2. extruder for raft results in GCode containing "G0 F3600 X0.00 Y0.00 Z0.75" with 2. extruder activated, crashing into 1. extruder in standby after raft is finished (see line 2638 in attached gcode)

### Actual results

Headcrash

### Expected results

No GCode containing command to move 2. extruder to 1. extruders parking position

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[headcrash.zip](https://github.com/Ultimaker/Cura/files/11754665/headcrash.zip)

GregValiant, 2023-06-15:
Thanks for the report.
Cura doesn't actually support IDEX printers.  It can be configured to work for some of them, but it is up to the user.  Cura assumes that both hot ends are mounted on the same carriage so they can't crash.  It also assumes that the left extruder cannot get to the right edge, and the right extruder cannot get to the left edge.
As a consequence this isn't really a Cura bug.  I'm going to remove the label but I'll leave this open so you can respond.

raspstar, 2023-06-15:
#ssc14100 #response_container_BBPPID{font-family: initial;font-size: initial;color: initial;} 😂Not the answer i expected but OK.S3D or postprocessing module.Case can be closed.Bye,Raspstar              Ursprüngliche Nachricht       Einzelheiten anzeigen                                                                                                                                                                         Von:  ***@***.***: 15. Juni 2023 15:52An:  ***@***.***:  ***@***.***:  ***@***.***; ***@***.***: Re: [Ultimaker/Cura] Using 2. extruder on IDEX results in headcrash after printing raft (Issue #15756)  
Thanks for the report.
Cura doesn't actually support IDEX printers.  It can be configured to work for some of them, but it is up to the user.  Cura assumes that both hot ends are mounted on the same carriage so they can't crash.  It also assumes that the left extruder cannot get to the right edge, and the right extruder cannot get to the left edge.
As a consequence this isn't really a Cura bug.  I'm going to remove the label but I'll leave this open so you can respond.

raspstar, 2023-06-15:
😂Not the answer i expected but OK.

S3D or postprocessing module.

Case can be closed.

[image]

closed 2023-06-16 by GregValiant