Ultimaker/Cura #15757: Error

cerberus09, 2023-06-15:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Creality Ender 3 S1

### Describe your model

Zoid model, have not been able to slice, atertight, ran through with meshtool plugin as well. 

### Add your .zip here ⬇️

[CE3S1_Berserk_Fury_EXE.zip](https://github.com/Ultimaker/Cura/files/11763566/CE3S1_Berserk_Fury_EXE.zip)

GregValiant, 2023-06-15:
Thanks for the report and the file.
Cura 5.3.1 won't slice this model.  I didn't find any errors in it and it should slice.

I'll pass this along to @MariMakes of the Cura team as it is one they may want to take a close look at.
I didn't have any trouble slicing it in 4.13.1 or PrusaSlicer.

MariMakes, 2023-06-20:
Hey @cerberus09,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for trying our beta and reporting your issue 👍

I'm getting an error that the Mesh is faulty. 🤔 
This is odd because you mentioned that you tried to fix it. 

Perhaps this could be some other tools you could try
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for.

closed 2023-07-04 by github-actions[bot]