Ultimaker/Cura #15758: Project slices to 0 minutes with no preview.

Gymnanji, 2023-06-15:
### Cura Version

5.3.1

### Operating System

Window 10 64

### Printer

Ender 3 S1

### Describe model location

No, I moved it all over.

### Describe your model

No, I've just retrieved the model from Thingiverse

### Add your .zip here ⬇️

[Newgearv4.zip](https://github.com/Ultimaker/Cura/files/11764002/Newgearv4.zip)
This project slices but shows a O minutes print time and no preview.  Possibly need to fix model but unsure as to how.

Asterchades, 2023-06-16:
The model is most definitely faulty. The only normals which don't appear to be reversed are in the superfluous, capped hollow in the middle, and the caps themselves which are perfectly overlapping other geometry that is reversed. As such the whole model is effectively inside out.

Cura has some facilities to deal with model faults but it isn't a modelling program so they're fairly limited. As a result it's treating the model as if it's been defined properly, and the definition it's been given has no possible way of being printed as the walls only have one side - two sides, pointed _away_ from each other, are required to generate a solid wall.

This is not a Cura issue but a modelling one. Ideally it would be corrected at the source but there are tools available for an end user to do so. The quickest, easiest, and least configurable (which can be a good or a bad thing) is the Microsoft 3D Builder tool which is free for Windows 10 users, and may already be installed on your system - if not it will be present in the Microsoft Store.

Gymnanji, 2023-06-16:
Ok, that was immensely helpful.  I pulled the model up in the Microsoft 3D Builder tool, it said it had an issue and asked if I wanted it repaired.  I selected 'yes', saved it and voila!  Thanks a bunch Asterchades!

Asterchades, 2023-06-17:
Outstanding. I'll close this one up as it sounds like it's been resolved.

closed 2023-06-17 by Asterchades