Ultimaker/Cura #15762: Model frame does not match model in z direction when Rift brim is activated

spaokD, 2023-06-16:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

Ultimaker 2 Extended+

### Reproduction steps

1. Load any 3D model into Cura
2. Enable Rift adhesion
3. Slice the model
4. Click on your model to activate the model framing (outer line of the 3D model)

### Actual results

Model is shifted up in z direction (due to Rift adhesion), but framing stays on original height.

### Expected results

Model and model framing are shifted up in height (z direction) simultaneous when Rift brim is used.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2023-06-16:
Thanks for the report.
The model is on the build plate at a Z of "0".  The bottom toolpath of the model is at a Z of (call it) 1.5.  This is expected behavior to show them both the way they are actually located.  I guess I can understand what you are saying but the model doesn't move up.
I'm going to remove the bug label from this as it is "expected behavior".  I hope you understand.  I'll leave this open for a bit so you can respond here if you feel that is premature.

spaokD, 2023-06-20:
Understood, thank you for the fast explanation. Just for me as a beginner, it seemed to be off in preview mode, since the shape is not matching the model anymore.

GregValiant, 2023-06-20:
The model is the model so the toolpaths do actually match the outline of the model.
The way they display together they are off by "Height of Raft + Raft Air Gap" though.
I'll go ahead and close this then.  Thanks for the report.

closed 2023-06-20 by GregValiant