Ultimaker/Cura #15767: Slicing fails when tree supports selected

chrisschnabel, 2023-06-16:
### Cura Version

5.3.1

### Operating System

Ubuntu 22.04.2 LTS

### Printer

Monoprice Maker Select Plus

### Name abnormal settings

Support Type = Tree (works for normal)

### Describe model location

Yes, when rotated to side, slicing does not fail.

### Describe your model

The slicing works unless support type = tree, in the configuration as loaded from stl.

### Add your .zip here ⬇️

The attached design does not slice when support type = tree. If it is rotated with support type = tree, it will slice. If it is support type normal, it will slice.

[Delta Dust Chute 900025 V1.1.zip](https://github.com/Ultimaker/Cura/files/11775817/Delta.Dust.Chute.900025.V1.1.zip)

chrisschnabel, 2023-06-16:
Original step file available here: https://cad.onshape.com/documents/f50737f67745fbc14bddb27c/w/fcff1aa7f56bb194fb5486f6/e/246072fe9c26f68a52c3ee36?renderMode=0&uiState=648ccc43e4a27c25f992bcdc

chrisschnabel, 2023-07-26:
> Hey @chrisschnabel <https://github.com/chrisschnabel>,
>
> Welcome to the UltiMaker Cura Github 🚀
> Sorry, it took us a while to get back to you 😞
>
> The new Cura 5.4 contains the fix for this for a number of slicing crashes.
> I did not have the exact printer settings that you have, but I was able to
> slice your model successfully in 5.4 Stable.
> [image: image]
> <https://user-images.githubusercontent.com/40423138/253968356-3ac8d5a2-1618-41ea-ad42-947dc4148320.png>
>
> Could you test if that's the case for you as well?
> You can download it here:
> https://github.com/Ultimaker/Cura/releases/tag/5.4.0
>
> If you still run into a crash, please double-check if you shared the
> correct file and let us know.
> I'll take another look at what is going wrong. 👍

This works now with the new version! Thank you!

MariMakes, 2023-07-27:
Great! I'll close this issue.

closed 2023-07-27 by MariMakes