Ultimaker/Cura #15769: Elegoo PLA Material Profile

IsaacUtah1379, 2023-06-16:
### Is your feature request related to a problem?

There doesn't seem to be a set of material profiles for Elegoo filament when choosing the material. There is also no longer an option for generic PLA.

### Describe the solution you'd like

Preferably, a set of material profiles for Elegoo filament would be added.

### Describe alternatives you've considered

Alternatively, it could also be useful to have the generic PLA profile put back on the list of options.

### Affected users and/or printers

The printer I've experienced this problem with is a Creality Ender 3 Pro. I haven't tried it with any other printers.

Th3Alic3, 2024-05-19:
Update on status of Cura adding
Elegoo 1KG +/- 0.02mm tolerance filament
material profile?

guiiids, 2024-06-13:
Also interested.

BigSpice, 2024-12-27:
>>>??
>Any action?

mdrokz, 2025-02-07:
Im also interested, any updates on this ?