Ultimaker/Cura #15771: Missing generic material profiles in 5.4.0 beta 1

HeadClot, 2023-06-17:
### Is your feature request related to a problem?

Currently there are a bunch of missing generic material profiles in 5.4.0 beta 1
The only generic material profiles in 5.4.0 beta 1 are Generic BVOH and Generic ASA.

### Describe the solution you'd like

I would like to see the various other generic material profiles added back to Cura in the next beta release if possible.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

It effects everyone that uses the generic material profiles that are not "Generic BVOH" and "Generic ASA".

### Additional information & file uploads

Material Profiles in 5.4.0 beta 1
[image]

Material profiles in 5.3.1

[image]

GregValiant, 2023-06-17:
Thanks for the report.
I found this to be true as well.  My printer is an Ender 3 Pro and within the new definition file is this:
        "exclude_materials": [
             .............
            "filo3d_pla_green",
            "filo3d_pla_red",
            "generic_abs",
            "generic_bam",
            "generic_cffcpe",
            "generic_cffpa",
            "generic_cpe",
            "generic_cpe_plus",
            "generic_gffcpe",
            "generic_gffpa",
            "generic_hips",
            "generic_nylon",
            "generic_pc",
            **"generic_petg",
            "generic_pla",**
            "generic_pp",
            "generic_pva",
            "generic_tough_pla",
            "generic_tpu",
It seems to me that no "generic" materials should be excluded, but this is in the Creality definition file so I'm not sure who did this.
The Cura team will take a look at this I'm sure.  I'm going to change this from a "new feature request" to a bug report.  I hope you don't mind.

HeadClot, 2023-06-18:
Hey @GregValiant - Thank you for the reply. Sorry for putting this under the feature request. I was tired when making the Bug report. Anyway thank you for looking into it :)

codewar65, 2023-06-18:
Using Generic BVOH in leau of Generic PLA. Getting best print I've had in months on my Ender 3 S1 Plus.

MariMakes, 2023-06-18:
Hey @HeadClot,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍 This looks like a serious issue 😨 

I'll bring it up to the team to see what they can do to improve it before the stable release.
Fingers crossed 🤞

fmontagut, 2023-06-18:
Hi @MariMakes thanks. Might not be related but definitively not the same error. I can see the generic materials in the list. If I add a custom printer, with same parameters as the ELEGOO printer, every works, including the generic materials, as intended.

<img width="1552" alt="Screenshot 2023-06-18 at 21 59 59" src="https://github.com/Ultimaker/Cura/assets/8792652/b39bfbeb-dd
[image]
af-4dba-9901-703897331345">

fmontagut, 2023-06-18:
second screenshot
[image]

fmontagut, 2023-06-18:
with a ELEGOO printer created as "custom", materials and slicing works properly:
[image]

fmontagut, 2023-06-18:
same screenshot using a custom configured printer. Generic materials work properly:
[image]

HeadClot, 2023-06-18:
Hey @fmontagut what version of Cura are you on?

fmontagut, 2023-06-18:
[image]
version: 5.4.0-beta.1

HeadClot, 2023-06-18:
Ok so some more information - I am using a Sovol SV06. It seems like in the excluded material definitions for the SV06 profile include the materials I talked about in my first post. Which is why the Neptune 3 Pro is not effected.

apexad, 2023-06-19:
Creality CR-10 is also affected by this bug

GregValiant, 2023-06-19:
@apexad The issue is in the "creality_base.def.json" file so all Creality printers are effected.  You can remove the "excluded materials" of your choice from that file and they should appear the next time you start Cura.

apexad, 2023-06-19:
@GregValiant on a Mac where is that file located?

GregValiant, 2023-06-19:
@MariMakes will be able to help with that.

MariMakes, 2023-06-20:
Hey @apexad,

I'm not a mac user myself, but I've troubleshooted this before. 
I believe these are the steps you need to follow. 

1. [Open Terminal](https://www.howtogeek.com/682770/how-to-open-the-terminal-on-a-mac/)
2. Type or copy paste
`open ~/Library/Application\ Support/cura`
3. Open 5.4 folder
4. Open the Definitions folder

You should be able to find it there. 🤞 
If it's not there, please let us  know and we'll try to find someone with an actual mac 💪

fieldOfView, 2023-06-20:
The `creality_base.def.json` definition is not in the configuration folder as pointed out by @MariMakes, but inside the app bundle. Ctrl-click the Ultimaker Cura icon and select "show package contents". Then look around in that folder for a folder named `definitions`. My mac is not working at the moment so I cannot tell the exact location.

Robenc, 2023-06-23:
i have the same issue with my ender 3 pro missing the same stuff and sovol has the same issue
[image]

GregValiant, 2023-06-23:
For the Sovol the problem is in the "sovol_base.def.json" file.  As stated, for Creality the problem is in "creality_base.def.json".  I'm sure this will be fixed for the stable release.  In the meantime you can manually edit those files in a text editor.

Raven-XIII, 2023-06-24:
@GregValiant & all, pain in the @ss but temporary fix/workaround. Open Cura 5.3.1 & export profiles needed then manual import to Beta. Like I said a pain but a lot quicker than a txt editor.

MariMakes, 2023-07-05:
Quick update from our side. 

This should be resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If it's not resolved for you, please leave a comment and we can reopen it. 👍

closed 2023-07-05 by MariMakes

TheNvisible, 2023-07-07:
HI MariMakes, 
I am having the same issue with the non-beta release. MAC OS 13.4.1, Elegoo N3 Pro. 

Appreciate the team's work; this is a great app!

TheNvisible, 2023-07-07:
[image]
[image]
[image]

MattPerron, 2023-07-07:
I have the same issue on my Elegoo Neptune 2S

apexad, 2023-07-07:
Creality CR-10 still not listing Generic ABS or Generic TPU which as far as I am aware it is capable of printing.

GregValiant, 2023-07-07:
It looks like the Creality_base.def.json file was not updated for 5.4.0.  Creality owners will need to edit that file and delete the lines for the materials they want included in Cura.
The Elegoo_base.def.json file appears to be correct in 5.4.0.  It no longer has PLA or PETG listed as disabled.
I don't have a definition file for a Neptune 25 so I can't check further.  If Elegoo handled it like Creality then the base file should be the only place the materials are addressed.

apexad, 2023-07-08:
> It looks like the Creality_base.def.json file was not updated for 5.4.0. Creality owners will need to edit that file and delete the lines for the materials they want included in Cura.
> The [...]

So, actually I had placed a `creality_base.def.json` file in `/Users/my-username/Library/Application\ Support/cura/5.4/definitions` and it was not being loaded by the beta but it seems like 5.4.0 was loading it.  I have removed it and I now see all the generics I could ever want:

[image]

From my standpoint this issue is resolved.

T3rm1, 2023-08-17:
I can confirm that Neptune 3 Pro no longer works with stable 5.4.0. Same problem as @TheNvisible

MariMakes, 2023-08-28:
We indeed are seeing some reports for people with Elegoo printers. 
We are collecting them here https://github.com/Ultimaker/Cura/issues/16007, but we have some trouble reproducing the issue on our side. There is some work done and there are some workarounds available.

So I highly suggest checking this thread out https://github.com/Ultimaker/Cura/issues/16007

wakandan, 2023-12-26:
@MariMakes the problem still persist in version 5.6.0. I'm using sovol sv06 plus
@apexad can I download the file from somewhere?

wakandan, 2023-12-26:
[image]
clicking on `Create New` does not do anything. And this is the latest stable release? What is it supposed to do then?