Ultimaker/Cura #15772: Configuration not supported error after adding Neptune 3 Pro

fmontagut, 2023-06-17:
### Application Version

5.4.0-beta.1

### Operating System

MacOs Ventura 13.4

### Printer

Elegoo Neptune 3 PRO

### Reproduction steps

1. I installed Cura
2. Deleted all settings (as there was an error with my custom materials)
3. Add a non-ultimaker, no-networked printer

### Actual results

No matter which generic material I choose, the "there are no profiles matching the configuration of this extruder" appears

### Expected results

the combination printers configurations-generic materials-default profiles should work by default

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11777988/cura.log)

ingoiot, 2023-06-18:
this is true for creality machines too(ender3 pro),
anycubic  (i3mega) does work.
(win10)

it seems they handle profiles differently.

MariMakes, 2023-06-18:
Hey @fmontagut,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Could this be a duplicate of https://github.com/Ultimaker/Cura/issues/15771, where all generic materials seem to be missing?
I'll bring it up to the team to see what they can do to improve it before the stable release.

closed 2023-07-03 by github-actions[bot]

T3rm1, 2023-08-17:
Same problem with final version of Cura 5.4.0. Generic profiles are visible but it still says that the configuration (in my case Generic PLA) is not supported.

MariMakes, 2023-08-28:
Hey @T3rm1,

Are you still stuck? We've provided a workaround here: 
https://github.com/Ultimaker/Cura/issues/16007