Ultimaker/Cura #15775: slice error.  it works with 5.3.1

tokuro, 2023-06-17:
### Cura Version

5.4.0 b1

### Operating System

win11

### Printer

ender 5

### Describe your model

fan cover.  it worked fine with 5.3.1

### Add your .zip here ⬇️

[cover.zip](https://github.com/Ultimaker/Cura/files/11778606/cover.zip)

GregValiant, 2023-06-17:
Thanks for the report and the file.
I can't duplicate this as your project sliced fine for me.
Have you tried moving the model around on the build plate and/or rotating the model?
The Cura team will take a look.

PaintOnBrush, 2023-06-18:
I fixed my issue which also sliced in 5.3.1, yet not in 5.4.0b1, by disabling "connect top/bottom polygons"