Ultimaker/Cura #15780: failed slice in beta

Barneq, 2023-06-17:
### Cura Version

5.4.0 beta.1

### Operating System

Windows 10

### Printer

solvol 3d sv04 single mode

### Name abnormal settings

Slicing tolerance changed from middle to exclusive at the time but others also am trying to get cura to stop filling the top of the vase

### Describe model location

Centre

### Describe your model

A vase

### Add your .zip here ⬇️

[failed slice.zip](https://github.com/Ultimaker/Cura/files/11778947/failed.slice.zip)

GregValiant, 2023-06-17:
Thanks for the report and the file.
You can see that Cura has painted a pattern on the inside of your model.  That means there are problems with the model and even if it was to slice it probably wouldn't be correct.  In this case it appears that the model is a surface and not a solid.
[image]

This is sliced in "Surface" mode.  You can see that these extrusions are so far apart that this would fail quite early even though the gcode would be correct.  Spiralize will not work for the same reason.
[image]

I'm going to remove the bug label as in it's current form this model just won't work with FDM.  Let us know if you feel that is premature.  I'll leave this open for a bit so you can respond.

Barneq, 2023-06-17:
Thanks Greg, I only logged the report because Cura wanted to, so thanks for the quick response. It is a new vase I'm working on & I was just playing around seeing how Cura would deal with it. It did a pretty good job using spirilise. Ultimately it has been a win because logging the report got me to sign up to Ultimaker & discover plugins for Cura which has got me very interested. If you know of any resources in regards to writing plugins for Cura I would love to check them out. Thanks in advance & have a great day.

[image]<https://user-images.githubusercontent.com/64202104/246618930-406def52-bb93-452d-a258-0d83c50a0c2f.png>

[image]<https://user-images.githubusercontent.com/64202104/246619188-8282be6e-b4d2-4612-be48-fe43f709b192.png>

GregValiant, 2023-06-18:
There are two kinds of plugins.  There are those that are "Extensions" and you will find them in the MarketPlace.  After installing them the files will be in the Configuration folder under Plugins.  Perusing them will give you and idea of the structure and how the Python scripts interact with the QML scripts that contain the textboxes, buttons, and other controls.
The other type are Post Processing plugins.  They are single file Python scripts that pull the raw gcode from Cura, make alterations, and then pass it back.  I have written a few post processors that I think will be good additions to Cura.  I still haven't got a good handle on doing Pull Requests though so we'll see.  They are in the "...share/cura/plugins/postprocessing/scripts" folder".  
There is also some information in the Cura Wiki that you can access from the icon up above on this page.

Barneq, 2023-06-18:
Thanks again Greg I'll jump in & have a look right now. Have a great weekend.