Ultimaker/Cura #15784: Slicing Failed after adding an additional Model

zentaur1ger, 2023-06-17:
### Cura Version

5.4.0 Beta 1

### Operating System

Windows 11

### Printer

Anycubic I3 Mega (modified)

### Describe model location

no

### Describe your model

this projekt contains two parts. the one that looks like a fan duct breaks the slicing. the other model slices fine

### Add your .zip here ⬇️

[EXTRUDER MOUNT.zip](https://github.com/Ultimaker/Cura/files/11779808/EXTRUDER.MOUNT.zip)

GregValiant, 2023-06-18:
Thanks for the report and the file.
I didn't have any trouble slicing your project so I'm unable to duplicate the problem
Have you tried moving the fan duct thing to different places on the build plate?

I made some changes.  You can scale support blockers like you can any other mesh so you really only need two.  Using the Per Model settings you can disable "Enable Supports" on the fan duct.

I'll leave the bug label on this and the Cura team will take a look.
Here is a project file with some changes.  It may slice but you may have to try moving the models around to fool the bug.

[gvEXTRUDER MOUNT.zip](https://github.com/Ultimaker/Cura/files/11782214/gvEXTRUDER.MOUNT.zip)

zentaur1ger, 2023-06-18:
Hi Greg,

yes I did. Turned it, changed the position, boring helped.
What I've noticed after sending the bug, after removing the duct from the
buildplate, with the other part still in place without moving (which sliced
successfully as long as it was the only part on the plate) this one also
fails.
Today I started my computer, put the same parts on the plate, and it sliced
without issues.
But it's worth noticing, that I've not sliced that project, but the same
parts, so the same files were used. Also I've implemented the support
blockers completely new.
Do I don't think, that the parts or the blockers are necessarily the
problem, maybe something in the program in general, that is gone after
restart. Something like a cache or so.

Kind regards

Oliver

zentaur1ger, 2023-06-19:
> I think this one is very close to what we see in other slicing errors of
> good models. One of the problems I have with these preliminary
> investigations is that I often can't duplicate the problem with my
> computer/printer combo. It makes trouble-shooting tough.
> Even when the problem turns out to be a bug (and they do) it is my
> personal preference to give the reporting person a workaround so they can
> print. It's no fun to have to wait for 3 months for the next release of
> Cura to fix a problem.
> In your case it isn't too bad because you found out that one model at a
> time will work so you aren't suffering from a Show Stopper. That isn't
> always the case.

Yes i'm indeed fine for now.
If you need some assistance (not code or such, but test something maybe)
feel free to reach out.

GregValiant, 2023-06-19:
Alright, here ya go.
Since you are on Windows here is a download link to "[Greg's Toolbox](https://drive.google.com/uc?export=download&id=1rntbmjRc4MLaDo7F2DbN2BwZWjBAJ8iJ)".  It's a full app for communicating with the printer.  It is a "Front End" rather than a "Print Server" (think Pronterface without PrintRun).  It (should) give full access to the printer from a PC.  It doesn't print via the USB but it will control printing via the SD card from a connected PC.  It's been checked with Creality 8 and 32 bit boards and has been fine.  There is a readme file in the zip folder.  It requires a USB data connection.  The Cura port drivers work fine for me.

Next is Greg's PostProcessors.  Unzip it and copy the files to your Cura configuration folder (mine is "C:\Users\...myname...\AppData\Roaming\cura\5.4\scripts".  My versions are named differently so you will be able to tell the difference when you load a post processor in Cura.  There is a readme PDF file in the zip.  I have been using some of these post processors for a while and haven't found any bugs lately.  A couple (like "Mullti-Extruder Color Mix") could use a second set of eyeballs as I only have a virtual printer to check it on.
[gv_PostProcessors.zip](https://github.com/Ultimaker/Cura/files/11792602/gv_PostProcessors.zip)

I write these things for myself and so I try to put in all the bells and whistles.  It boils down to "Oh joy!!!  Another learning curve."

Enjoy.

zentaur1ger, 2023-06-20:
As i'm running an Anycubic I3M, i've also klipperized it.
So it only works with klipper.
If i've understood anything right, all this works only with marlin,
correct?

GregValiant, 2023-06-20:
It sends Gcodes.  If they are passing through Klipper---I don't know.  If it sends G28 the printer should auto-home.
There are specific Klipper commands like the PAUSE script that are NOT included.  The "Pause" script in the program uses M25. 
 Whether that will work or not I don't know.
You would need to send that manually using the box on the Communication tab.
I'll take a look at Klipper and see what commands it uses.  I could add a check box to switch from Marlin to Klipper.  I would have to change some things like using PAUSE instead of M25.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic