Ultimaker/Cura #15785: Failed Slice with every print i try.

NateslPlayzzz, 2023-06-17:
### Cura Version

5.4.0 beta

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Name abnormal settings

No, Will not slice anything at all with supports or no supports.

### Describe model location

Centre

### Describe your model

Base for Iron Man Bust by Wicked 3d Prints

### Add your .zip here ⬇️

[Bust Base.zip](https://github.com/Ultimaker/Cura/files/11779881/Bust.Base.zip)
[image]

NateslPlayzzz, 2023-06-17:
Tested on 5.3.1 and it slices fine

Zenith816, 2023-06-17:
I noticed the same thing whenever I use my 5.3 custom settings..  With the stock profiles it normally works..

MariMakes, 2023-06-18:
Hey @NateslPlayzzz,

Thanks for your report 👍

Could this be a duplicate of https://github.com/Ultimaker/Cura/issues/15797?
Connect Top/Bottom Polygons seem to be broken, do you have that setting enabled in your custom profile?

If that's not it, I need a little bit more information. 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting, you shared an STL instead. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

NateslPlayzzz, 2023-06-18:
Yes, The Top/Bottom Polygons are connected, however i tried disabling them yet it still fails slicing. I believe this similar to a recent one now where Custom Settings is not working. Would google drive work to transfer the file, or do i specifically need to use WeTransfer? If so, what is the email?

NateslPlayzzz, 2023-06-18:
nvm found the link thing.

NateslPlayzzz, 2023-06-18:
For the heck of it i tested it one last time, and it did slice, It might of been the top and bottom polygons after all, so sorry for the confusion! Thanks for the help!

MariMakes, 2023-06-19:
Hey @NateslPlayzzz,

Thanks for the update and double-checking the setting!
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15797
This issue has been added to the list to resolve before the stable release.

Thanks for testing our Beta and reporting the issues.

closed 2023-06-19 by MariMakes