Ultimaker/Cura #15790: item failed to slice after adding anti warping tabs

wluckie1, 2023-06-18:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

creality cr-10

### Name abnormal settings

i added anti warping tabs

### Describe model location

i have rotated model 90 degrees to print upside down. model sliced fine before adding the tabs

### Describe your model

basic model which is part of laser pistol

### Add your .zip here ⬇️

[AEP7_laser_pistol_Frame_front_brace.zip](https://github.com/Ultimaker/Cura/files/11781178/AEP7_laser_pistol_Frame_front_brace.zip)

5axes, 2023-06-20:
Same issue as : https://github.com/Ultimaker/Cura/issues/15807  

Looks like an issue with the Mesh type Support : 
[image]

MariMakes, 2023-06-21:
Hey @wluckie1,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for using the beta and reporting your issues👍 

You are not the first person to report this, I'll mark this issue as a duplicate of https://github.com/Ultimaker/Cura/issues/15836
I'm bringing it up to the team to see what they can do to improve it before the Stable release,

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15836

closed 2023-06-21 by MariMakes