Ultimaker/Cura #15795: Will not slice when using custom profiles

Myklt8, 2023-06-18:
### Cura Version

5.4.0 beta 1

### Operating System

Windows 11

### Printer

Franky (Ender 3 modded)

### Name abnormal settings

Any custom, slices with default, even with changes, as long as I don't change the name

### Describe model location

no

### Describe your model

Yes it slices on 5.3.1, I did try the fix mesh

### Add your .zip here ⬇️

[CE3S1PRO_SN V1 IPA Cap - 1 Spout small.zip]
(https://github.com/Ultimaker/Cura/files/11782020/CE3S1PRO_SN.V1.IPA.Cap.-.1.Spout.small.zip)

Will not slice when using custom profiles. I can use the default and it slices. I can change settings in the default and it slices.
If I change the name from the default, it will not slice

MariMakes, 2023-06-18:
Hey @Myklt8,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I took a look and your issue seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15797, where it's reported that Connect Top/Bottom Polygons seems to be broken and it seems to be enabled by default in your profiles. 😢 

I'll bring it up to the team to see what they can do to improve it before the stable release. 
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15797

Thanks for trying our Beta and reporting your findings 😊

MariMakes, 2023-06-18:
Oh! Is this you? I love this make 😍
https://www.thingiverse.com/make:1111435

closed 2023-06-18 by MariMakes