Ultimaker/Cura #15797: Eror slice as connect Top/Bottom Polygons

ArhaNNN, 2023-06-18:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Ender 3 pro

### Name abnormal settings

Yes - connect Top/Bottom Polygons

### Describe your model

No

### Add your .zip here ⬇️

[CE3PRO_Fabulous Amur (3).zip](https://github.com/Ultimaker/Cura/files/11782361/CE3PRO_Fabulous.Amur.3.zip)

MariMakes, 2023-06-18:
Hey @ArhaNNN,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I can reproduce it, it seemed to work in Cura 5.3.1 but not anymore. Thanks for pointing us in the right direction. 

I'll bring it up to the team to see what they can do to improve it before the Stable Release.
Fingers crossed 🤞

MariMakes, 2023-07-05:
Thanks again for the report. 💪 
We were able to resolve this issue in the 5.4 Stable release. 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

I'll close the issue since it's resolved.

closed 2023-07-05 by MariMakes