Ultimaker/Cura #15798: Error Dialog on Start, Windows 8.1: From PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, The specified procedure was not found.TextEdit, QGroupBox, QCheckBox, QPushButton ImportError: DLL load failed while importing QtWidgets: Die angegebene Prozedur wurde nicht gefunden.

chiefstobs, 2023-06-18:
### Application Version

5.4.0.b1

### Operating System

Windows 8.1 64bit

### Printer

Ender 3 S1 Pro

### Reproduction steps

Start Cura 5.4.0b1 on Windows 8.1 64bit

From PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, The specified procedure was not found.TextEdit, QGroupBox, QCheckBox, QPushButton ImportError: DLL load failed while importing QtWidgets: Die angegebene Prozedur wurde nicht gefunden.

### Actual results

Start Cura 5.4.0b1 on Windows 8.1 64bit

From PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, The specified procedure was not found.TextEdit, QGroupBox, QCheckBox, QPushButton ImportError: DLL load failed while importing QtWidgets: Die angegebene Prozedur wurde nicht gefunden.

### Expected results

Cura should start like 5.3.0 X-Mas Alpha

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-06-18:
Hey @chiefstobs,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This is not an error I recognize, error, but I do know that Windows 8 is not on our Supported Operating Systems list.
You can learn more about which version is compatible with your OS here
https://support.makerbot.com/s/article/1667337917781

You are the first person I meet that could run a recent cura version on that version of Windows 8.
I believe what you are seeing is a change introduced here: https://github.com/Ultimaker/Cura/pull/14908

To me, it doesn't seem like a bug in Ultimaker Cura, since Windows 8 is not on our supported OS list. 
Would you still like to keep this issue open or do you agree that we can close it?

chiefstobs, 2023-06-19:
Hi @MariMakes 

thank you for your answer. I guess it is time to finally upgrade to Windows 10 / 11 :D 

The issue can be closed then I guess :)

MariMakes, 2023-06-19:
Hey @chiefstobs,

Good luck with the upgrade and happy printing!
I'll be closing this issue.

closed 2023-06-19 by MariMakes