Ultimaker/Cura #15800: Slicing Falure after windows 10 updates ran.

TheDanzilla, 2023-06-19:
### Cura Version

5.4.0-beta.1

### Operating System

Windows 10

### Printer

Creality Ender 3 v2

### Name abnormal settings

n/a

### Describe your model

Parts of a small robot figurine 

### Add your .zip here ⬇️

[CE3E3V2_EyeBrows.zip](https://github.com/Ultimaker/Cura/files/11782971/CE3E3V2_EyeBrows.zip)

MariMakes, 2023-07-17:
Hey @TheDanzilla,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
That's a pretty cool model 🤩

Even in Cura 5.4 I can reproduce your slicing crash. 
It doesn't seem to stop Slicing.

I'll ask one of the developers to take a look at this file when they resolve the slicing crashes. 

Thanks again, and please let us know if you run into any other issues 💪

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic