Ultimaker/Cura #15805: slicing failed using custom profile

kaiju50c, 2023-06-19:
### Cura Version

5.4.1

### Operating System

windows10

### Printer

ender 3/v2

### Name abnormal settings

it slices with default profile but not custom

### Describe model location

where it should be

### Describe your model

a cyinder

### Add your .zip here ⬇️

[Untitled.zip](https://github.com/Ultimaker/Cura/files/11786773/Untitled.zip)

TheSoumilArora, 2023-06-19:
I have the same issue on ender 3v2 neo
not able to use custom profiles for slicing
Also lost all my materials
Like it shows generic PLA and TPU are installed but not able to see them in material menu

closed 2023-06-19 by MariMakes

kaiju50c, 2023-06-20:
> Hey @kaiju50c.
> 
> Welcome to the UltiMaker Cura Github 🚀 Thanks for using our beta and reporting your findings.
> 
> I checked your project file and what you are seeing is already reported here: #15797 The setting Connect Top Bottom Polygons is broken in this beta, it's added to the list to be resolved before the stable release. If you disable the settings you can slice as usual. [image]
> 
> I'll be closing this issue as a duplicate but you can follow the progress here: #15797
> 
> @TheSoumilArora, what you are seeing seems to be related to #15771

thats explains alot. I am so noob. thanks!