Ultimaker/Cura #15806: Slicing failed after some time. At that time, it seems half of the job was done.

kleinlaut, 2023-06-19:
### Cura Version

 5.2.1 utt also 5.3.1

### Operating System

Windows 10

### Printer

Anycubic Vyper

### Name abnormal settings

"normal"

### Describe model location

"normal"

### Describe your model

Many holes in a plate,  created by OpenScad, Rendering took about 14h.

### Add your .zip here ⬇️

Just stopped after a while in Cura 5.2.1 and 5.3.1.
Hardware 12 Core, 32 GB RAM. Drive "C:\↕" had 11 GB free disk space when slicing started.
[AV_brunnenfilter_0.6_wrk_laufzeit_14Std.zip](https://github.com/Ultimaker/Cura/files/11787371/AV_brunnenfilter_0.6_wrk_laufzeit_14Std.zip)

AngusX74, 2023-06-20:
Hi,
I have a similar problem. Unlike mine, your model is sliced correctly with the default profile (in Cura 5.4.0 Beta1). So the problem seems to have something to do with your individual settings. For analysis it could be helpful if you experiment with the settings and check when the model is no longer sliced correctly.

kleinlaut, 2023-06-20:
Hi AngusX74,
I have been using the profile for all my PETG prints without any problems. Now I have done two things. First, I changed the OpenScad code so that it now takes about 2.5 hours to render instead of the previous 14 hours. Then I tried to render the result with my previous profile. This was unsuccessful. I then rendered it with another similar profile and that worked. I can now print a sample with my Vyper printer and if everything is ok with the resin printer the final part, you don't have to slice it with Cura. I am satisfied with this, thank you very much for your effort and your tip.

kleinlaut, 2023-06-20:
Changing the profile worked.

closed 2023-06-20 by kleinlaut