Ultimaker/Cura #15809: After update program repeatedly crashes loading ui.

TheHEadFool, 2023-06-19:
### Application Version

5.4.0

### Operating System

windows 10

### Printer

Ender 3

### Reproduction steps

When I star the program, it crashes after getting through the loading UI. Takes several attempts to get it to successfully launch. This started after the last update.

### Actual results

Program fails to open.

### Expected results

Should have opened.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

MariMakes, 2023-06-19:
Hey @TheHEadFool,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for using our beta and reporting your issues👍

I'm getting some flashbacks from https://github.com/Ultimaker/Cura/issues/6197, which took us a while to resolve 😓 
I've seen some slower boot-up times with my windows set-up as well, but I thought it was because I have some testing versions installed. 

I'll bring it up to the team to see what they can do to improve it.

In the mean time, I have some questions
1. Can you share if you see a splash screen when you start Cura?
2. Do you see a crash window when starting Cura?
3. What was the Cura version you used before upgrading? 
4. Can you start Cura in administrator mode?

TheHEadFool, 2023-06-19:
1. Can you share if you see a splash screen when you start Cura?

Yes a splash screen apears and shows as the components load up. After
loading UI. it disappears. Program does not always open - it takes a few
times.

   1. Do you see a crash window when starting Cura?

No, No crash screen.

   1. What was the Cura version you used before upgrading?

5.3.1

   1. Can you start Cura in administrator mode?

With the same results as in NON-Administrator mode.

cgolchert, 2023-06-19:
I get the crash window as soon as the splash screen goes away.  I never make it to any GL screen.  5.3.1 worked fine for me as well.

MariMakes, 2023-06-21:
Hey all,

The team took a look and they need some extra information for troubleshooting.

Would you mind sharing your log files? 
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log