Ultimaker/Cura #15814: Slicing failed after screen resolution change - Cura moved between two screens.

muschaw, 2023-06-19:
### Cura Version

5.4.0-Beta-1

### Operating System

MacOS

### Printer

Lotmaxx Shark

### Name abnormal settings

.

### Describe model location

.

### Describe your model

Yes, the model can be sliced.

### Add your .zip here ⬇️

[LS_X1_Spoolholder_1kg.3mf.zip](https://github.com/Ultimaker/Cura/files/11791432/LS_X1_Spoolholder_1kg.3mf.zip)

I suspect that this happened because I switched the screen resolution. 
I moved Cura between two slightly different scaling screens). One is my USB-C screen, and the other is my Intel Mac 15-inch screen.

As you can see in the screenshot, the preview window scaling also failed right after the screen move. 

[image]

muschaw, 2023-06-19:
I can reproduce the behaviour even without a model loaded (the preview scaling fails to scale back to the screen size).
moving Cura between windows and zooming in/out reproduces the problem every time.

muschaw, 2023-06-19:
Ok, I stumbled on two separate bugs 😅 

the screen resolution scaling is one, and the model not slicing is a separate one. 
this model does not slice in 5.4.0-Beta-1 

Should I open a separate issue?

muschaw, 2023-06-19:
[cura.log](https://github.com/Ultimaker/Cura/files/11792677/cura.log)

Can this be related to this export exit code? 
I don't mind enabling additional symbols / debugs if needed. 

2023-06-19 21:42:22,019 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-06-19 21:42:22.017] [info] Starting export...
2023-06-19 21:42:22,093 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [982]: Backend exited abnormally with return code -11!
2023-06-19 21:42:22,096 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [375]: Attempting to kill the engine process
2023-06-19 21:42:22,097 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [381]: Killing engine process
2023-06-19 21:42:22,097 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [384]: Engine process is killed. Received return code -11

muschaw, 2023-06-20:
This is from a different model - same setup : 

2023-06-20 22:38:59,939 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-06-20 22:38:59.937] [info] Starting export...
2023-06-20 22:39:00,038 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [982]: Backend exited abnormally with return code -11!
2023-06-20 22:39:00,041 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [375]: Attempting to kill the engine process
2023-06-20 22:39:00,041 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [381]: Killing engine process
2023-06-20 22:39:00,042 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [384]: Engine process is killed. Received return code -11
2023-06-20 22:39:00,042 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
2023-06-20 22:39:00,045 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Listening
2023-06-20 22:39:00,065 - INFO - [MainThread] UM.Backend.Backend.startEngine [95]: Started engine process: /Applications/UltiMaker Cura.app/Contents/MacOS/CuraEngine
2023-06-20 22:39:00,066 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [115]: [Backend] Calling engine with: ['/Applications/UltiMaker Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '']
[LS_Easythreed_X1_Controller_Holder001.3mf.zip](https://github.com/Ultimaker/Cura/files/11806512/LS_Easythreed_X1_Controller_Holder001.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11806518/cura.log)

muschaw, 2023-06-20:
it does slice the model with the default profiles. 
I tried disabling supports on my profiles - imported from 5.3.1 (as I'm using tree I figured it's worth a shot) but no dice. 

my config is uploaded in the previous comment - using the "slow good" profile.

MariMakes, 2023-07-17:
Hey @muschaw,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.4 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.4 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

MariMakes, 2023-07-24:
Great! Thanks for the update 😊 
I'll close this issue since you are not stuck anymore.

closed 2023-07-24 by MariMakes