Ultimaker/Cura #15816: No Nozzle option available for Neptune 3 Plus

Echo113, 2023-06-19:
### Application Version

5.4.0-beta.1

### Operating System

Windows 11

### Printer

Neptune 3 Plus

### Reproduction steps

Added Neptune 3 Plus from the "add printer". Ive tried uninstalling and reinstalling, but nothing works. The profiles worked fine on the last builds.
[image]
[image]
[image]
[image]
[image]
[image]

### Actual results

Nozzle shows empty under filament. Changing settings does nothing.

### Expected results

Same results

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I even tried coping a friends folder contents but nothing changed.

GregValiant, 2023-06-19:
Thanks for the report.
#15817 is a pull request for this.  There is a typo in your nozzle files in this line in each of the 4 files:
definition = elegoo_neptune_3max
should be:
definition = elegoo_neptune_3plus

MariMakes, 2023-06-20:
Hey @Echo113,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

@GregValiant found the typo and proposed a way to fix it. 
Hopefully we can introduce the fix for the stable release 🤞

Echo113, 2023-06-20:
> Hey @Echo113,
> 
> Welcome to the UltiMaker Cura Github 🚀 Thanks for your report 👍
> 
> @GregValiant found the typo and proposed a way to fix it. Hopefully we can introduce the fix for the stable release 🤞

Thanks! I applied the fix and it works now.