Ultimaker/Cura #15818: Cura 5.4.0 beta1; An extra skirt line on layer 1 and layer 2 (using layer 0 as bottom layer)

csterlin, 2023-06-20:
### Application Version

5.4.0 beta1

### Operating System

Windows 11 Home, 22H2, 22621.1778, Intel(R) Iris(R) Xe Graphics

### Printer

Creality Ende 3

### Reproduction steps

Sliced model. Exported to sd-card. Inserted card into printer. Selected pre-heat PLA. Printed model.
[image]
[image]
[image]

### Actual results

The print runs normally until the start of layer one (layer 2 in the jpeg) in which it adds a skirt line to the existing skirt. This happens again at the start of layer two (layer 3 in the jpeg), but runs normally after starting layer 3 (layer 4 in the jpeg).

### Expected results

There should be no additional skirt lines after layer 0 completes.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[HoseHolder_4.zip](https://github.com/Ultimaker/Cura/files/11794087/HoseHolder_4.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11794095/cura.log)

csterlin, 2023-06-20:
[HoseHolder_4.zip](https://github.com/Ultimaker/Cura/files/11794156/HoseHolder_4.zip)
Added gcode and stl files to zip archive

closed 2023-06-20 by csterlin

opjose, 2023-07-06:
I see the same issue with 5.4.0 non-beta

csterlin, 2023-07-06:
Turns out that there's a new feature, skirt height, which also affects the
corner tab+ plugin, and which has a workaround by setting skirt height to
one.

opjose, 2023-07-06:
Thanks!

Found it. Strangely it did not appear in the list until I searched for it even though I have all options enabled.
Even then it shows up slightly greyed out as if the setting is not applicable, yet it still works.

ThomasCura, 2023-08-23:
Hello! I also have the problem that the skirt is printed over several layers, even though I don't want to. Where is the switch that can be used to reduce skirt to 1? I haven't found anything.

ThomasCura, 2023-08-23:
Now I've also found him with the search. Greyed out. Setting it to 1 gives the desired result. Set to 3 again (default), the setting no longer works. It stays with one layer. Somehow the function is not implemented correctly. It works for me if only one skirt layer should be printed.