Ultimaker/Cura #15820: Cleaning distance of the outer wall

Khronos666, 2023-06-20:
### Application Version

5.x.x

### Operating System

Win10 

### Printer

Castom

### Reproduction steps

1. The cleaning distance of the outer wall... does not work when the wall is in one line,
in 2 lines before version 5.3.1 did not work ...
in version 5.4.0-beta works, but in the opposite direction... it reaches the beginning of the seam and is mirrored in the opposite direction ...
in 3 lines it already works as it should ...
2. the rolling option behaves the same as the cleaning distance....

raised the topic https://github.com/Ultimaker/Cura/discussions/14637#discussioncomment-5107549

1. Расстояние очистки внешней стенки... не работает когда стенка в одну линию, 
    в 2 линии до версии 5.3.1 не работала...
    в версии 5.4.0-beta работает , но в обратную сторону... доходит до начала шва и отзеркаливается в обратную сторону...
    в 3 линии уже работает как надо...
2. опция наката ведет себя так же как расстояние очистки....

тему подымал https://github.com/Ultimaker/Cura/discussions/14637#discussioncomment-5107549
[Cura 5.3.0-alpha_beta_.zip](https://github.com/Ultimaker/Cura/files/11796361/Cura.5.3.0-alpha_beta_.zip)

### Actual results

?

### Expected results

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

?

MariMakes, 2023-07-11:
This comment seems to be related: 

Version 5.4.0 has been released ... The bug was never fixed... damn, since version 5.3 alpha pointed out the incorrectness of this setting ...
I am glad that you are developing the project and introducing new functions, but again you are breaking already working ones ...
but this option is very useful...on some plastics, she glues the seam in thin walls ...
is it really only me who has such a bug on all versions of 5.x.x?

it works like this - https://github.com/Ultimaker/Cura/assets/123404675/a7f856ae-6efe-4ddc-8c62-63632c84e4a9

it doesn't work that way correctly - https://github.com/Ultimaker/Cura/assets/123404675/647478df-bbe7-4286-81c0-4ec5dda6ff3f

it doesn't work that way at all - https://github.com/Ultimaker/Cura/assets/123404675/16dc1fea-c4f8-4ec7-9c05-f8ff89aed9d6

MariMakes, 2023-07-11:
Hey @Khronos666,

Thanks for your first issue in Cura🚀 
Sorry, it took us a while to get back to you 😞 

I do recall discussing this issue with colleagues before. It seems to be an issue that only shows up with only a very limited amount of walls that have random Z-seams. 

Can you help me understand why this feature is important to you?
What kind of problem are you trying to solve? 
Do you have any pictures of how this is impacting your printed part quality?

Khronos666, 2023-07-14:
The tasks are different, let's say printing a cylinder in one wall or a vase in one wall....
Why is this option critical? It is critical for Watson plastic based on SBS polymer...
it is viscous...and without rolling, the seam does not close...there is a hole in the seam...
there is no such problem on other plastics....since they are more fluid....

on version 5.x.x
[image]

on version 4.x.x
[image]

MariMakes, 2023-07-17:
Ah, what you are seeing is related to thin-walled structures. 
You might be running into problems with the new Arachne Slicing Engine. 

​​​​​​​There are settings that you can change to improve this behavior.
You can read more about them here: 
https://support.ultimaker.com/hc/en-us/articles/4792077687068

Can you take a look to see if there are settings here that can help you out?

Khronos666, 2023-07-22:
It's not from slicing...and not from the new settings... this is from a slightly broken algorithm...  I even cut a video of how CURA simulates movement, everything is very clearly visible where it works and where it doesn't... everything is very simple, the roll does not work in either one wall line or two wall lines...and with three or more it works perfectly...  

You can score on this post...the version is already 5.4, and the bug has remained as it was... we'll have to put up with it... or in some projects go back to version 4.x.x ...
Thank you for your attention ...)))
Спасибо за внимание...)))

MariMakes, 2023-07-24:
🤔  Oh! Could this be a duplicate of https://github.com/Ultimaker/Cura/issues/15279?
There they also talk about gaps in thin walls.

Khronos666, 2023-07-30:
> 🤔 О! Может ли это быть дубликатом #15279? Там также говорится о зазорах в тонких стенах.

no... but it looks like... I have no problem with the PEGT...it is fluid and does not need rolling ...
but Watson based on SBS polymer does not stick together well...therefore, it peels off, the situation improves if you strengthen the airflow, but then the part is stratified... therefore, rolling along the Z seam was just necessary...

Look carefully, CURA herself shows how she generated GCOD ...
Attention to the line of movement, the blue line is the roll of smoothing the Z seam....

on this video it works without complaints

https://github.com/Ultimaker/Cura/assets/123404675/1b09c8df-6a65-459a-912b-44749337f615

in this video, it works with the opposite effect

https://github.com/Ultimaker/Cura/assets/123404675/42f72f3c-ce43-42a9-b49b-aeef344ad758

it doesn't work at all on this video

https://github.com/Ultimaker/Cura/assets/123404675/a0d686f9-5f0c-427d-86a9-3610b882ad9a

The error is not in the printer, nor in the plastic, the GCODE generation error...

It is enough to turn on the display of the movement of the extruder... and you will see how the model will be printed...

all...With that, I take my leave, I have nothing more to say to you... the error just beats in the eyes, but you don't want to notice it in any way...

it remains to wait, suddenly accidentally fix ...)))

Khronos666, 2023-10-13:
finally, a new version of UltiMaker Core 5.5.0-beta has been released.1
But this bug has not changed ...
the roll is also missing...
:(

Hal9999, 2024-04-06:
can't confirm on 2 lines case: seems to be working as expected:

https://github.com/Ultimaker/Cura/assets/15959100/4948a59a-26b8-480d-9f7f-52c2b7634309

Zogar89, 2024-05-10:
I found the problem, setting Wall Ordering = Inside to Outside makes te problem. 
Outside to Inside solve the problem.

Khronos666, 2024-05-10:
> В Cura 5.6 в случае с 2 строками, похоже, все работает так, как ожидалось:
> 
>  2024-04-06_22-27-26_ультимейкер-Cura.mp4

Yes, indeed, it works if the wall printing order is from exterior to interior... but it doesn't work in one line anymore...

Khronos666, 2024-05-10:
> I found the problem, setting Wall Ordering = Inside to Outside makes te problem. Outside to Inside solve the problem.

this is not a solution, it's just a workaround, besides, it doesn't work in one line anyway...