Ultimaker/Cura #15824: 5.3.1 / 5.4.0-beta1 - Missing brim / skirt

grbd, 2023-06-20:
### Application Version

5.3.1 / 5.4.0-beta1

### Operating System

Windows 10

### Printer

Anycubic Kossel Linear Plus

### Reproduction steps

I figured I'd open a new issue since the original one was closed down
and this seems to affect both 5.3.1 and 5.4.0-beta1
https://github.com/Ultimaker/Cura/issues/14876

Basically in certain circumstances part of the brim or skirt will be missing on a print
I've put an example 3mf project file here:
https://github.com/Hecatron-Cad/Tools.3DPrinter.Anycubic.Kossel/blob/master/Schematics/Upgrades/Voron-M4/AKLP_end_plate.3mf

You can work around it by dragging the part to a different area such as to the left or right
Maybe it's related to the fact I'm using a delta? I'm not sure

### Actual results

Missing brim / skirt

### Expected results

brim / skirt

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

GregValiant, 2023-06-20:
Thanks for the report and the link.
You can Zip a 3mf file and post the zip folder here.  I think there is a 25mb limit on the zip folder size.

This looks like a bug to me.  I'll guess and say that the brim is being cut off in a response to the way Cura now allows models to be closer to the edge of the build plate by trimming the skirt/brim.  This one just looks weird as the part extends beyond the trim line.

@MariMakes may want to take a look at this one.

grbd, 2023-06-20:
[AKLP_end_plate.zip](https://github.com/Ultimaker/Cura/files/11801067/AKLP_end_plate.zip)
Ok now zipped and uploaded
I've noticed it does impact the g-code so isn't just a visual artifact, also it affects both brim and skirt.

MariMakes, 2023-06-20:
Hey @grbd,

We discussed https://github.com/Ultimaker/Cura/issues/14876 with the team already, but I probably haven't updated the ticket correctly. We can see in our data that the combination of these two settings Enable Coasting + Wipe Nozzle Between Layers are quite rare, and that the issue disappears if one of them is disabled. 
Is disabling one of the two settings an option for you? 

We believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I'll close this issue, I hope you understand.

closed 2023-06-20 by MariMakes

GregValiant, 2023-06-20:
I see the model has some errors as well.
Here you can see that the project model has the funky brim.  The model at the bottom was repaired with MS 3D Builder and the brim is what we would expect.
[image]

grbd, 2023-06-20:
As far as Coasting + Wipe Nozzle Between Layers those features are already disabled
I don't have anything Experimental enabled

I tried importing the stl into MS 3D Builder but it's saying everything is ok

[image]

grbd, 2023-06-20:
[end_plate.zip](https://github.com/Ultimaker/Cura/files/11805445/end_plate.zip)
This is the stl file for info

grbd, 2023-06-20:
Also had a quick go with meshlab just in case it was something that 3D Builder wasn't seeing
But the results were the same
https://www.youtube.com/watch?v=IreaoCfQ8mE&t=83s

I should mention though that moving to a different area of the build plate will give you different results as far as what part of the brim / skirt is masked off

This is fine:
[image]

over here it gets weird
[image]

GregValiant, 2023-06-20:
This is with the AnyCubic Kossel active.
[image]

And this is with an Ender 5 active.
[image]

This is an FLSUN 400
[image]

Every time I switch back to the Kossel the effect reappears.
I noted that in the original project file neither "Enable Coasting" nor "Wipe Nozzle Between Layers" were enabled.
I installed a Vyper and it didn't have the effect either.
This is what I get in MS 3D Builder.
[image]

@MariMakes this only seems to occur with the AnyCubic Kossel, but those other two settings weren't involved.

GregValiant, 2023-06-20:
With a 110mm brim it gets weirder.
[image]

[Brim problem.zip](https://github.com/Ultimaker/Cura/files/11806000/Brim.problem.zip)

GregValiant, 2023-06-20:
Alright, here is the deal.
The Kossel Linear Pro inherits the regular Kossel definition.
Both definition files have "Disallowed Areas" defined.  Those disallowed areas are causing Cura to act like it would if the brim was falling off the edge of the build plate.
I think the disallowed areas are simply written incorrectly in the definition files.
I have to guess here, but it seems like the intent was to disallow all of the circular build plate except for a 230 x 230 square in the center of the build plate.

This is back to square one, with the original project file BUT with the "Disallowed Areas" deleted from the definition files.
[image]

And that Ladies and Gentleman deserves a beer.