Ultimaker/Cura #15826: Faulty slicing: random yellow slicing in random places unexpectedly

goofyseeker311, 2023-06-20:
### Application Version

5.3.1

### Operating System

Win11 RTX3060

### Printer

elegoo neptune 3 pro

### Reproduction steps

Slicing a whoop drone with large duct drums (4x circular tubes).
Raft adhesion, no supports, only vertical features with horizontal 4mm holes.

### Actual results

One of the tubes had a thin yellow layer in the middle vertically, while other duct tubes were clear.

### Expected results

All the tubes should be clear (sometimes are, after a second slicing of the same model with same settings).

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11801192/cura.log)
[EN3PRO_tailsitterdroneplane-v64-chassis.3mf.zip](https://github.com/Ultimaker/Cura/files/11801197/EN3PRO_tailsitterdroneplane-v64-chassis.3mf.zip)

GregValiant, 2023-06-21:
Thanks for the report and the link.
I'm not sure what the "yellow line" could be.  The model has a continuous outer wall and if you have the "starts" turned on in the color scheme dialog then that's what you could be seeing.  There would be only one external Z seam on that model.
If you go to the Walls Settings and make the Zseam X start = 0 and the Zseam Y start = 0 and then check the box for "Zseam Relative" it should hide the seam and keep it off the diameters.
You have Coasting turned on and the Preview only shows moves that are extrusions.  As a consequence there will be gaps in the outer walls that wil allow the interior to show through.  Turn coasting off just to check the preview to see if it is a display thing.

goofyseeker311, 2023-06-21:
I don't know, it's like one of the circles has a middle level "yellow eye"
plane (not line), extrusion that should not be there.

goofyseeker311, 2023-06-21:
A side question: should the "remove support" button do anything? seems not to be functioning at all, or in any intuitive way.

GregValiant, 2023-06-21:
To add a support blocker the model you want the blocker for has to be selected first, then click the blocker button, and click somewhere inside the blue edge of the model.  I usually pick something right at the edge so I can see it better.
Blockers can be moved and scaled like any other mesh.  They can also have different settings than the model (see the Per Model options).
Post a screen shot of the "yellow line" problem.  I'm having trouble visualizing it.

goofyseeker311, 2023-06-22:
cant get it to reproduce, even after multiple re-slicing of the same shape. here is an artistic similarity paint of the what happened.

[image]

GregValiant, 2023-06-22:
Nice job on that.  It looks like a model error was encountered during the slice.  I checked the model and it doesn't appear to have any errors so I'm not sure what happened there.

Since I can't reproduce it, and you can't reproduce it, I vote we close this (but leave the bug label on it).  It can be revisited if it happens again.
The Cura team must be able to reproduce a fault in order to chase a bug down.  Let me know what you think.

goofyseeker311, 2023-06-22:
as I remember, it happened when first time ever touching the original .stl (slicing) file with the cura slicing, then it fixed itself after toggling support and re-slicing.

GregValiant, 2023-06-22:
I'll go ahead and close this then.  If it happens again let us know and this can be re-opened or you can create a new report.

closed 2023-06-22 by GregValiant

goofyseeker311, 2023-06-24:
another similar weird issue, is when you slice a model, then for the first time slide the layer slider on the right, it will begin on the bottom, then do nothing but raise a grey bar to the top, then start working properly, ie, the image does not match where/what there slider is supposed to be showing, visualization mismatch.