Ultimaker/Cura #15832: Load new project crashes Cura

mikejbare, 2023-06-21:
### Application Version

5.4.0 Beta 1

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro 

### Reproduction steps

I loaded a STL into Cura then saved the project and exited Cura.   I then restarted Cura and used File / Open to read the 3mf file I just created.

### Actual results

When I restart Cura and attempt to open the project I just save, Cura crashes.   The same thing happens if I drag and drop the 3mf file onto Cura.  This problem is repeatable.

### Expected results

-

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[pinhead.zip](https://github.com/Ultimaker/Cura/files/11810760/pinhead.zip)

GregValiant, 2023-06-21:
Thanks for the report.
I can't duplicate this with your file as it opens fine for me.  I did have this issue when I first installed 5.4b.
With Cura running go to the Materials and see if "Generic PLA" is available.  It ended up on the Excluded Materials list for Creality printers and if your "Tweaked PLA" is based on the Generic PLA it might be causing the problem.

MariMakes, 2023-06-21:
Hey @mikejbare,

Thanks for your report 👍

This reminds me of behavior that happens when you close Cura while it's still saving. Your project file becomes corrupted without proper warning. Could this be the case for you as well?

Do you mind repeating your steps to reproduce but with a small shape [like a cube](https://www.thingiverse.com/thing:1278865)?

mikejbare, 2023-06-22:
Simple Cube
[image: Screenshot 2023-06-22 155501.png]
I don't slice or anything, just save the project

[image: Screenshot 2023-06-22 155546.png]
I exit Cura then click on the project file I just saved

[image: Screenshot 2023-06-22 155636.png]

Here is the result:

[image: Screenshot 2023-06-22 155701.png]

> Message ID: ***@***.***>
>

GregValiant, 2023-06-23:
Sorry @mikejbare but using email doesn't allow attachments to come through on GitHub.  The images need to be posted directly here.
The best would be a Project File.  Load the cube, use the "File | Save Project" command, zip the 3mf file and post the zip folder here directly - not via an email.

mikejbare, 2023-06-23:
Ok, I will try.

mikejbare, 2023-06-23:
To the developer who thought that my profile was based on a profile from a previous version of Cura that doesn't exist in the current, beta version, and that was his/her idea of why Cura was crashing when attempting to open a 3mf file.   Here's why I think you are correct:
  I create a new custom profile by duplicating one of the existing profiles for my printer, "Standard Quality".   Now when I load my STL, change the profile to "Standard Quality #2", select "Generate Support" and "Tree", when it slices, it's perfect, nothing extraneous, nothing unsupported, perfect.  You should probably have a more graceful way of checking if the profile is OK or not but I don't think this is a show stopper.

GregValiant, 2023-06-23:
I have come across problems like that.  When they are brought to the Developer's attention they get fixed but it's like any bug,  if you are unaware it exists then you can't fix it.  Someone has to bring it up and point it out.  I know when Arachne was rolled into mainstream Cura that several of these "Gotchas" were fixed ahead of time.  There were settings that disappeared from the 4.x versions that still had to be handled by the 5.x version.  Were they all caught?...probably not.  Cura is a complex piece of software. 
 It's an ongoing battle.

So it's important that users make these reports because otherwise "...the software is perfect".  HA!!

MariMakes, 2023-07-21:
Hey @mikejbare,

Quick update from our side 👋
I showed the team your issue and they were really curious. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10834

Thanks again for the report! 👍

GregValiant, 2023-11-14:
@MariMakes  I have another one here.  This is from bug report #17312 >> [Project Zip](https://github.com/Ultimaker/Cura/files/13348526/FV_GENERIC.MALE.-.Camlock.Male.Connection.v0.zip)

From the crash dialog Upper text box:
Traceback (most recent call last):
  File "cura\CuraApplication.py", line 1163, in event
    return super().event(event)
  File "UM\Qt\QtApplication.py", line 502, in event
    event._function_event.call()
  File "UM\Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "cura\Utils\Threading.py", line 34, in _handle_call
    ico.result = func(*args, **kwargs)
  File "C:\Program Files\UltiMaker Cura 5.5.0\share\cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 857, in read
    self._applyChangesToMachine(global_stack, extruder_stack_dict)
  File "C:\Program Files\UltiMaker Cura 5.5.0\share\cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 1188, in _applyChangesToMachine
    self._applyVariants(global_stack, extruder_stack_dict)
  File "C:\Program Files\UltiMaker Cura 5.5.0\share\cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 1162, in _applyVariants
    node = ContainerTree.getInstance().machines[global_stack.definition.getId()].variants[variant_name]
KeyError: '0.4'

And from the lower text box:
Thread 0x00000690 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000006fc (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\UltiMaker Cura 5.5.0\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002768 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004180 (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.5.0\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000043b0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000049ac (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 600 in wait
  File "cura\Utils\Threading.py", line 39 in _call_on_qt_thread_wrapper
  File "UM\Workspace\WorkspaceFileHandler.py", line 42 in readerRead
  File "UM\FileHandler\ReadFileJob.py", line 78 in run
  File "UM\JobQueue.py", line 133 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000457c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001430 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002124 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002bd8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002280 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000043a4 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00002ad4 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 165 in exceptHook
  File "UM\Qt\QtApplication.py", line 416 in exec
  File "cura\CuraApplication.py", line 925 in run
  File "cura_app.py", line 239 in <module>

Cura.log
[cura.zip](https://github.com/Ultimaker/Cura/files/13348855/cura.zip)