Ultimaker/Cura #15833: ARM / Apple Silicon / Modern Apple Machine Support (M1/M2 CPUs)

sammcj, 2023-06-21:
### Application Version

All

### Operating System

macOS 13.4

### Printer

Ultimaker S7

### Reproduction steps

It seems that both the current stable and beta versions of Cura are still only build for Intel / x86_64 architecture.

This causes significant performance issues as it has to be run in the Rosetta emulator on macOS.

Given that ARM / Apple Silicon has been out for 3 years now and all modern Apple machines are no longer Intel based - Cura should be built for native ARM.

Cura is the only intel-only binary still running on my machines and the differences is _very_ noticeable.

If Cura's CI system cannot build ARM packages, it would be incredibly helpful if the build scripts / instructions worked for ARM so users could optionally build their own packages.

### Actual results

Cura runs in emulated Intel x86 mode with poor performance.

### Expected results

Cura to run in native mode, not emulated.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is somewhat related to #11841 with the addition that it's not only beta builds - but the production stable builds that aren't created properly.

cc/ #8795 and probably others.

TheSin-, 2023-08-10:
M1 build of 5.4.0, for others that are waiting, the thread if very long so this link will go directly to it.

https://github.com/Ultimaker/Cura/issues/11841#issuecomment-1669888233

jellespijker, 2023-08-18:
I'm in the process of setting up the workflows and hooking up the physical self-hosted runner for building Cura for the M1.

**Can people try the binaries in this run, which are compiled for the native M1's:** https://github.com/Ultimaker/Cura/actions/runs/5906663040

- https://github.com/Ultimaker/Cura/suites/15242781431/artifacts/871059202
- https://github.com/Ultimaker/Cura/suites/15242781431/artifacts/871059203

I am mostly curious if these run on older MacOSes

gordio, 2023-08-19:
_I am mostly curious if these run on older MacOSes_
At least on newest (Sonoma M1) — works.

closed 2023-08-23 by casperlamboo