Ultimaker/Cura #15834: [5.4.0 Beta1] Fuzzy Skin Bug

Zohan-4775, 2023-06-21:
### Application Version

5.4.0 Beta1

### Operating System

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

Fuzzy Skin Outside Only Checked

### Actual results

Parts (some small lines) at the Outside without Fuzzy Skin

### Expected results

Fuzzy Skin everywhere at the Outside

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura_Bug.zip](https://github.com/Ultimaker/Cura/files/11811571/Cura_Bug.zip)
[image]
[image]

MariMakes, 2023-06-21:
Hey @Zohan-4775,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for trying our beta and reporting your issue👍

I can reproduce it on your model, it happens when the holes on the side start and end. 

It does seem model specific because it does not happen on my model 🤔 
[image]

I'll bring it up to the team to see what they can do to improve it, or at least explain what is happening. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

Zohan-4775, 2023-06-21:
Hi @MariMakes 
Drill some Horizontal Holes to you Model and see what happenes ;-)

MariMakes, 2023-06-26:
Hey @Zohan-4775,

One of our testers took a look and confirmed the issue, also that it's present in previous releases. 
The problem seems to happen here, with the single wall. 
[image]

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10729

Thanks for the report! 👍

Tad009, 2023-12-10:
Issue is still here in 5.6.0
[image]