Ultimaker/Cura #15836: custom support issue

Scull215, 2023-06-21:
### Application Version

5.4.0

### Operating System

windows

### Printer

elegoo neptune 3 plus

### Reproduction steps

I placed Custom supports.

### Actual results

i added custom supports and it failed to slice

### Expected results

it should be albe to slice with custom supports

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[EN3P_clamp base.zip](https://github.com/Ultimaker/Cura/files/11813999/EN3P_clamp.base.zip)

MariMakes, 2023-06-21:
Hey @Scull215,

Thanks for trying our beta and reporting your issues 👍
As @5axes correctly noted, you are not the only one reporting these issues.

I can succesfully slice your models with a build from a week before the release, sos something must have changed in that period. 
[image]

I'll bring it up to the team to see what they can do to improve it before the Stable release,
Fingers crossed 🤞

MariMakes, 2023-06-22:
Quick update,

We believe to have found the root cause of the bug. 
It should be resolved here: https://github.com/Ultimaker/CuraEngine/commit/1022fe09ef59d4d572da60f30fa6225c6a36fefe, that means we can probably include a fix in the 5.4 stable release. 

Sorry for the inconvenience! Thanks for the report! 💪

MariMakes, 2023-07-05:
This issue is resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

I'm glad we were able to resolve this for the stable release. 😍 
Thanks again for the report, we would have missed it without the report 💪

closed 2023-07-05 by MariMakes