Ultimaker/Cura #15837: Support blocker is not enforced, there is a support right at/in the block volume.

goofyseeker311, 2023-06-21:
### Application Version

5.3.1

### Operating System

Win11 RTX3060

### Printer

elegoo neptune 3 pro

### Reproduction steps

tilt a long print a bit to the side, so that it has a long support at the side, place a support blocker at the support.

### Actual results

the slicer will place a support in the support blocker location despite there is a support blocker in place.

### Expected results

There should not be a support in the support blocker position.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

any long print, like a 2mm sheet, titled 15deg so that its a bit partially sideways, so that support mode turned on generates a vertical column support at some position on the vertically longest distance part of the print.

MariMakes, 2023-06-21:
Hey @goofyseeker311,

Thanks for your report 👍
I don't believe Support Blockers indicate area's that no support is allowed, they indicate the red overhang areas that it overlaps with should not receive any support. 

Are you using Tree Support? If you use the new 5.4 beta, the tree supports behave differently. They will actually avoid the mesh like you mentioned.

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

goofyseeker311, 2023-06-21:
no, just standard supports, not tree supports.

MariMakes, 2023-06-26:
Hey @goofyseeker311,

Thanks for the files👍
I don't believe you shared the correct 3MF file, but I was able to reproduce your issue. 

I did notice that when I resized the support blocker with the scale tool, I was able to prevent the behavior from happening all together. 
[image]

Here is the project file I used: [Mari_EN3PRO_ultimakercurasupportslice.zip](https://github.com/Ultimaker/Cura/files/11866831/Mari_EN3PRO_ultimakercurasupportslice.zip)

This really doesn't seem like a bug to me. 
So I'll close the issue, I hope you understand.

closed 2023-06-26 by MariMakes