Ultimaker/Cura #15841: Supports does not support anything

StmLord, 2023-06-21:
### Application Version

5.4.0 beta 1

### Operating System

MacOS Ventura 13.4

### Printer

Custom FFF

### Reproduction steps

Added an object that need supports, I selected supports everywhere, concentric support and 1 perimeter wall on supports.

### Actual results

The result was a perimeter support around the needed supporting brunch but without supporting it at all.
Also in every other support type, it still creates a perimeter that is not on the same thickness of the brunches and goes around starting from build plate instead of the lower brunch.
[image]

### Expected results

Expected result is supports that the perimeter are starting on the lower brunch and don't get out of it at all

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

included 3mf project file
[support problem.3mf.zip](https://github.com/Ultimaker/Cura/files/11816647/support.problem.3mf.zip)

GregValiant, 2023-06-21:
Thanks for the report and the file.
Turn on the Custom settings and make adjustments to the Support settings.  The way it is set up now is not appropriate for this model.
This is with "Grid" support pattern and with the Support Interface enabled at 50% density.  The parts are well supported.
I'll remove the bug label as this is a setting issue.  I'll close this as well.  I hope you understand.
[image]

closed 2023-06-21 by GregValiant

StmLord, 2023-06-23:
is there a way for supports to not exceed the width of the supported object? in 4.3.1 the setting "supports everywhere" were no more than the width of the supported object and that way supports would not start from buildplate.

GregValiant, 2023-06-23:
There are 73 support settings.  When you hover the mouse over a setting you should get a popup of what the setting does and what other settings are affected by it.  There is a Plugin in the Marketplace that will enhance those popups to give a more detailed description.
If the Support Horizontal Expansion is > 0 then the supports will be wider than the print.
This is with Support Horizontal Expansion = 2
[image]

This is Support Horizontal Expansion = 0
[image]