Ultimaker/Cura #15843: Сталкер в Экзоскелете фигурка

AniDUB2901, 2023-06-21:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

Ender 3

### Describe model location

В любом положении

### Describe your model

Обыкновенная фигурка на полку

### Add your .zip here ⬇️

[фигурка экзоскилет.zip](https://github.com/Ultimaker/Cura/files/11816820/default.zip)

GregValiant, 2023-06-21:
Thanks for the report and the file.
This is a view in Mesh Mixer.  The red surfaces have errors and there are a lot of them.
[image]

I tried to repair the model but Cura still would not slice it.

This is the barrel of the gun in PrusaSlicer.  This won't survive removing the supports.
[image]

I'll leave the bug label on this and someone from the Cura team will take a look.

MariMakes, 2023-07-05:
Welcome to the UltiMaker Cura Github 🚀 

I agree with @GregValiant; This mesh seems faulty.
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for.

I'll close the issue, but if you run into any other issues feel free to create a new bug.

closed 2023-07-05 by MariMakes