Ultimaker/Cura #15846: Summary of configuration parameters in Gcode

KOPY4, 2023-06-21:
### Is your feature request related to a problem?

The problem arises when one needs to know what configuration parameters were used when creating the Gcode file.

### Describe the solution you'd like

The solution that I suggest is that at the end of the lines of the printing priority, in the Gcode, leave all the configuration parameters, in the same order that they appear in the printing parameters, one parameter, one line, something that is readable.
Another improvement, these summary lines may be useful to import them for a new project.

### Describe alternatives you've considered

The solution that I suggest is that at the end of the lines of the printing priority, in the Gcode, leave all the configuration parameters, in the same order that they appear in the printing parameters, one parameter, one line, something that is readable.
Another improvement, these summary lines may be useful to import them for a new project.

### Affected users and/or printers

Affects the user-print interface

GregValiant, 2023-06-21:
This will be proposed.  Unzip it and put the .py file in the ".../plugins/postprocessingplugins/scripts" folder and it will be available as a Post Processor.  I had to do some filtering so let me know if there are settings you feel I missed.  
[gv_AddCuraSettings.zip](https://github.com/Ultimaker/Cura/files/11824468/gv_AddCuraSettings.zip)

GregValiant, 2025-12-09:
My proposal for Add Cura Settings" was denied as being too hard to maintain.  I agree with that.
There are some profile settings that are added by "DisplayInfoOnLCD".
```
;FLAVOR:Marlin
;TIME:4709
;Cura Time Estimate: 1hr 18min 
;Extruder 1 (T0)
;  Filament used: 12.34m
;  Filament Type: PETG
;  Filament Dia.: 1.75mm
;  Nozzle Size  : 0.6mm
;  Print Temp.  : 235°
;  Bed Temp.    : 75°
;Electric Cost: $0.03
;Filament Cost: $0.94
;Layer height: 0.24
;Initial Layer Height: 0.24
;Base Quality Name  : 'Standard Quality'
;Custom Quality Name: 'GV PETG'
;Model List: ['Jug Cap.stl']
```

In addition in "Manage Profiles" you can "Import" a gcode file and Cura will rebuild a Profile based on the setting changes logged at the end of the gcode file.

I'll go ahead and close this.  I hope you understand.