Ultimaker/Cura #15849: This is happening regulaily.

Rich-1d12, 2023-06-22:
### Cura Version

5.4 beta

### Operating System

Windows 11

### Printer

Eryone Thinker SE

### Name abnormal settings

Unchecked Use Adaptive Layers.Sliced fine prior.

### Describe model location

All over the bed

### Describe your model

Yes sliced fine before turing off adaptive layers.

### Add your .zip here ⬇️

[ETS_Dragon_Egg___Solid.zip](https://github.com/Ultimaker/Cura/files/11828851/ETS_Dragon_Egg___Solid.zip)

fieldOfView, 2023-06-22:
_What_ is happening regularly?

The template forgets to ask you what your actual problem is, and unfortunately there is no way for us to tell.

Rich-1d12, 2023-06-22:
Sorry. It's Slicing failure errors.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic