Ultimaker/Cura #15850: Quick Guide popups in Machine settings Start and End G-Code keeps popping open and closed

MichaelCain2, 2023-06-22:
### Application Version

5.4.0 Beta.1

### Operating System

Mac OS Ventura 13.5 Beta

### Printer

Ender 3, 3+ and 5

### Reproduction steps

While editing the Start and end G-Code in Machine Properties, the popups keep popping up and and closing, continuously and have been even in last version 5.3.0. Might be a plugin if it is not part of the Cura feature menu help items.

https://github.com/Ultimaker/Cura/assets/44505092/a71e3821-d747-49a9-8056-9e743a237ad1

### Actual results

See recording

### Expected results

Menu comes up once, stays for few seconds unless I interact with it and if I do nothing, it closes until I choose the start or end g-code area, after exiting that area.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

MariMakes, 2023-06-22:
Hey @MichaelCain2,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for trying out our Beta and reporting your issue👍

I'm not sure, but it looks like a bug in the Settings Guid instead of the one in Cura. 
Would you mind trying the workaround mentioned in this issue: https://github.com/Ultimaker/Cura/issues/14963#issuecomment-1476118032?

Can you let us know if that resolves what you are running into?
If that's not the case I'm happy to take another look!

MichaelCain2, 2023-06-22:
Thank you @MariMakes I will check it out this evening. I appreciate you

MariMakes, 2023-07-10:
Hey @MichaelCain2,

Did you have a chance to take a look? 
|

closed 2023-07-25 by github-actions[bot]

acydic, 2024-12-15:
Tool tips plugin causes this behavior. Turn off in plugin settings.