Ultimaker/Cura #15852: missing volumes

iguvendikler, 2023-06-22:
### Application Version

5.4.0-beta

### Operating System

windows11

### Printer

ender3 pro

### Reproduction steps

[image]
I sliced my datas on cura for ender 3 pro and 3 v2. While on cure, every thing looking good and normal.

### Actual results

Ass you see the attached image.  Printer make this. some volumes are remove or axis are wrong after a while. 
[image]

### Expected results

[image]
I expected left one on picture.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

first time I have experience this issues for over 3 years.

GregValiant, 2023-06-22:
Thanks for the report.
Please provide a project file by loading the model (just the bad one) and setting Cura up, and then use the "File | Save Project" command.  Zip the 3mf file and post the zip folder here.
Without a project file I have to guess and I'd say that looks like a bad model file to me.  It might have a duplicate surface or two.  I've seen that happen before.

MariMakes, 2023-06-22:
Hey @iguvendikler,

Thanks for the update.
Your pictures scared me a bit, but I'm glad to hear your issue has been resolved.
I'll close the issue, feel free to let us know if you find anything new 😊

closed 2023-06-22 by MariMakes