Ultimaker/Cura #15855: Slicing failed with Extension Cylindric custom support applied

Gerhard-59, 2023-06-22:
### Cura Version

5.4.0-beta.1

### Operating System

Windows 10 64 bit

### Printer

Anycubic Kobra

### Name abnormal settings

Modell gets sliced without Custom Cylindric Support applied, but failes if one CCS is applied.

### Describe your model

Model of an old transistor radio

### Add your .zip here ⬇️

[AK_Radio Philips Original N250522.zip](https://github.com/Ultimaker/Cura/files/11833306/AK_Radio.Philips.Original.N250522.zip)

The slicer crashes when a Custom Cylindric Support is applied to the model. See alos enclosed screenshot

[image]

GregValiant, 2023-06-22:
Thanks for the report and the file.
The highlighted "polka dot" areas are added by Cura when there are errors in the model.  That model needs repair.
Here is a view in Mesh Mixer.  All those colored balls indicate errors.
I'll leave the bug label on this for a bit, but that model needs help.
[image]

Gerhard-59, 2023-06-22:
Issue can be closed from my side, with the fix planed for the stable release.

closed 2023-06-26 by MariMakes