Ultimaker/Cura #15867: Tap Anti warping plugin

Gadjet5, 2023-06-22:
### Application Version

5.4.1 beta

### Operating System

macos ventura

### Printer

kp3s

### Reproduction steps

added Tab Anti Warping discs (plugin) to corner of parts

### Actual results

Fails to slice using the Tab Anti Warping plugin

### Expected results

has previously sliced without issues 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

TabAntiWarping plugin version 1.4.0

GregValiant, 2023-06-22:
Thanks for the report.
Set Cura up like you had it and use the "File | Save Project" command.  Zip that 3mf file.  Use the "Help | Show Configuration Folder" command and locate the Cura.log file and add it to the zip folder.  Post the Zip folder here.

The plugin comes from a member of the community and @5axes will need more info to investigate this.

5axes, 2023-06-22:
No,  problem comming from Cura .. Nothing to do with the plugins.  But hopfully already solved by the Ultimaker Team : https://github.com/Ultimaker/Cura/issues/15836#issuecomment-1602573073

MariMakes, 2023-06-28:
👋 Hi!

@5axes is correct. It was an issue on the Cura side and it's resolved for the stable.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15836

closed 2023-06-28 by MariMakes