Ultimaker/Cura #15872: STOP ASKING ME TO WIPE ALL MY SETTINGS

sabikazad, 2023-06-23:
### Application Version

5.3.0

### Operating System

WIndow 10

### Printer

CE5 and CE3v2

### Reproduction steps

every time i open the program
i am asked if i want to wipe all my settings
stop
or give me the option to tell it to stop every time
so simple
probably less than 20 lines of code

### Actual results

add a don not show again box to the "COnfiguration Error" report...

why would i want to accidentally wipe my settings and profiles by clicking that every time the program opens

### Expected results

stop asking me to wipe my setting sand profiles every time i open the app its ridiculous

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
my configuration works great
i will be pissed if i accidentally click reset because i cant get that to stop popping up everytime i open this

Asterchades, 2023-06-24:
If you don't correct the issue, why would you expect it to stop reporting it?

The Cura log (which you didn't post) will show where it's encountering errors in your config files. You can fix the affected file/s using any regular text editor, at which point not only will the profile work properly (despite your claims that it already does) but it will also stop warning you that it's invalid.

Should you be unable to find the appropriate entries in the log file, post it here and someone else will be able to find it for you.

sabikazad, 2023-06-24:
It can be reported without having the only button be the clear your
settings. And like I said no issues with my prints or settings.
Great ui/ux team!

MariMakes, 2023-06-28:
Hi @sabikazad,

Welcome to the UltiMaker Cura Github 🚀 
I run into this issue myself as well sometimes.

One of our community members has written a plug-in that can help remove this message. 
It's the Start-Up optimizer and you can download it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/StartOptimiser

I'll bring it up to the team to see what they can do to improve it without the plug-in.
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

MariMakes, 2023-06-28:
Hey @sabikazad,

We discussed the issue with the team. 
It's something we encounter often as well, and this report inspired us to try to actually do something about it. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10739

Thanks for the report! 👍