Ultimaker/Cura #15873: Missing option check box

kabaltah, 2023-06-23:
### Application Version

5.4.0 beta1

### Operating System

windows 10

### Printer

Ender3 s1 pro

### Reproduction steps

"Support horizontal expansion" check box is not shown on the menu even in all or expert mo0des are selected.
 "Support interface horizontal expension" options are available but they don't work without the support horizontal expansion option clicked.

### Actual results

Compared the menu to version 5.3.0, missing option checkbox is available in older versions.

### Expected results

This option  should have been opeden via options menu but it does not.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Support horizontal expansion button is not available in version 5.4.0 beta1,  though it should.

fieldOfView, 2023-06-23:
I think that is because the combination of support horizontal expansion and the new tree supports leads to problems, so support horizontal expansion is disabled when the Support Structure is set to "Tree". Try setting the support structure to "Normal".

kabaltah, 2023-06-23:
@fieldOfView , Hello I had been using xmas special edition with tree supports since it had been released with great success. This option was enabled in that version. Also it is shown on the filter options menu in the new version, but it does'nt exist in the options menu.

MariMakes, 2023-06-28:
Hey @kabaltah,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Support Horizontal Expansion is a setting that is not supposed to be applied to tree supports, that is what other settings like Trunk, Branch and Tip diameter are for. But by mistake, this setting was not disabled correctly.

For the 5.4 stable, we have removed the setting for tree support correctly 
https://github.com/Ultimaker/CuraEngine/pull/1890

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

kabaltah, 2023-06-28:
Hello, @MariMakes , I exclusively print miniatures. There are many vertical conic overhangs on those models. Horizontal expansion is very usefull in those cases. 
 When tree supports expand trough sides they cover the overhang like a glove, so the supports doesn't fail for narrow highly vertical overhangs. 
 I use xmas special edition of cura since it is released , printed many miniatures using new tree supports any horizontal expansion is a life saver. 
 Is there a good reason the remove this feature from the new releases?

kabaltah, 2023-07-06:
@MariMakes  Hello, 5.4 staple has been released but support interface horizontal expansion buttons still exist. Also the are totally useless without support horizontal expansion option which is still missing. I will be better if you either totally remove the feature or implement the support horizontal expansion and make that option usefull again.

MariMakes, 2023-10-11:
👋 Quick update on our side. 

We made some improvements in the 5.5 Beta release.
With this fix https://github.com/Ultimaker/Cura/pull/16421 the warning value won't interfere anymore. 
[image]

You can download the Cura version here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes