Ultimaker/Cura #15877: Hole Horizontal Expansion Max Diameter not showing in "Print settings"

3DPrintingGeek, 2023-06-23:
### Application Version

5.4.0-beta.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 2

### Reproduction steps

1. Activate "All" in "Print settings", to make all settings visible.
2. Expand the "Walls" section. 
3. Look for the setting "Hole Horizontal Expansion Max Diameter".
4. The setting is not showing up.
5. Open "Manage settings visibility...".
6. Search "Hole Horizontal Expansion Max Diameter".
7. Trying if unchecking and checking it changes something -> it doesn't. Setting still not visible in "Print settings".

### Actual results

Setting for "Hole Horizontal Expansion Max Diameter" is not accessible in 'Print settings'.

[image]

### Expected results

The setting should be accessible and tunable in "Print settings" in the "Walls" section, when activated in "Manage settings.visibility..."

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

na

MariMakes, 2023-06-28:
Hey @3DPrintingGeek,

Welcome to the UltiMaker Cura Github 🚀 
Excited to see you here 😊 

[I took a look a look at the definition](https://github.com/Ultimaker/Cura/blob/6565295af47dff450da817011e6c8e47fc53a382/resources/definitions/fdmprinter.def.json#L1309C24-L1309C24) and the setting only shows up if the Hole Horizontal Expansion is enabled as you can see here:
[image]

The setting behaves similarly to settings like Minimum Feature Size that only show up once Thin Walls is enabled.
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

3DPrintingGeek, 2023-06-28:
Now that I know how it works it could be closed, though without knowing that the setting visibility depends on another setting is irritating. 
Maybe there's a way to improve that? What do you think?

MariMakes, 2023-07-10:
The Hole Horizontal Expansion Max Diameter setting will not do anything unless the Hole Horizontal Expansion is enabled, that setting is only enabled if the value is not equal to 0, so by not making the setting available, we can prevent people from changing it while it does not do anything.

Another thing that I did notice is that the Affects/Affected By is missing in this tooltip while I believe it's super applicable.
Here you can see what that looks like. 
[image]

I'll ask the team if we can update the description to include this extra information.

I'll close this issue since it's working as expected. But I'll update you if we end up updating the tooltip. 
Thanks for your report 👍

MariMakes, 2023-07-21:
Hey @3DPrintingGeek,

With these new insights, it resulted in quite the team. 
We decided to resolve the confusion in two ways. 

One is updating the tooltip for Hole Horizontal Expansion to reflect the relation with Hole Horizontal Expansion Max Diameter
The other is restoring the tooltip informing you of why a setting is missing. 

This I icon seems to have gone missing since the 5.0 release for Windows users, a significant part of the developers don't work on Windows machines so they weren't aware it was broken.
[image]

We've added a ticket to the backlog with the intent to restore the missing tooltip. 
For internal reference CURA-10835

Thanks for the report! 👍

MariMakes, 2023-09-01:
Good news. 😊 

It looks like this icon with the explanation of why settings are hidden will also be restored in the next release. 

[image]