Ultimaker/Cura #15880: Missing materials , no pla or petg

Robenc, 2023-06-23:
### Application Version

5.4 beta

### Operating System

windows 10

### Printer

Ender 3 pro

### Reproduction steps

missing attached materials for each prints it across multiple machines not just creality ender 3 pro 

### Actual results

materials not listed

### Expected results

a list of all generic materials

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

faradayberry, 2023-06-24:
Same is happening for me. Seems to be most but not all creality printers missing almost all generic materials. If you go to manage materials and search generic PLA it is installed but the creality profile won't show it.

Windows 10

This issue does not happen in the 5.3 release.

slabua, 2023-07-01:
Same here, all my custom materials are gone. Linux modern version.

MariMakes, 2023-07-05:
Hey @Robenc,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I recognize this issue. 🕵️‍♀️
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15771, where we made an error by excluding generic materials that should always show in printers.  

It's resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
I'll close this issue since it's a duplicate and resolved. 

Feel free to create a ticket if you encounter more issues, together we can make Cura better 💪

closed 2023-07-05 by MariMakes