Ultimaker/Cura #15884: Tree Support does not print on 1st layer.

anubis71x, 2023-06-23:
### Application Version

5.4.0 beta

### Operating System

Win 10

### Printer

v400

### Reproduction steps

Tree Support does not print on 1st layer.

Unchecking "enable support brim" resolved the issue.

Support interface was on and its effect on this issue was not tested, but that was my next thought.

### Actual results

When "enable support brim" is checked supports are printed starting on layer 2 and they fail.

### Expected results

When "enable support brim" is unchecked, supports behave normally.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

?

MariMakes, 2023-07-05:
Hey @anubis71x,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Could this be a duplicate of https://github.com/Ultimaker/Cura/issues/14983?
Where support brim is missing if buildplate adhesion is set to None.
We have a ticket on our backlog with the intent to improve this behavior.

Can you confirm if this is the issue you are running into? 
If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

anubis71x, 2023-07-05:
Changing build plate adhesion type also resolved my issue, so it may indeed be a duplicate. Works with a skirt, but not with "none".

MariMakes, 2023-07-11:
Thanks for the update 👍 

We have a ticket on our backlog with the intent to improve this behavior. (hopefully soon 🤞 )
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14983

closed 2023-07-11 by MariMakes