Ultimaker/Cura #15885: Change Filament at x.x

Abardenn, 2023-06-23:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

I3 Mega S

### Reproduction steps

1. Go to Extensions > Post Processing > Modify G-Code
2. Click Filament Change & Insert 0.x into layer (x being any number of your choice from 1-9)
3. Slice 

### Actual results

No M600 gets sliced into the G-Code. Which I'm not sure wether that's a bug or just not supported yet in Cura. This is what is also showed in the cura.log which I'm not sure if it's related

```
2023-06-24 04:00:02,445 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.addScriptToList [240]: Adding script ChangeAtZ to list.
2023-06-24 04:00:02,452 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///C:/Program Files/UltiMaker Cura 5.3.1/share/cura/resources/qml/Settings/SettingTextField.qml:161: TypeError: Cannot read property 'type' of null
```

### Expected results

M600 gets added to the G-Code at the correct decimal height

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11851433/cura.log)

MariMakes, 2023-07-05:
Hey @Abardenn,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Do I understand correctly that you want to insert a filament change at a layer number that is not an absolute number? 
So Layer 0.5 instead of layer 1? I don't believe Cura accepts these values. 

Can you help me understand why you don't want to enter an absolute number here?

MariMakes, 2023-07-05:
Also, if that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-07-20 by github-actions[bot]