Ultimaker/Cura #15886: Minimum Support Area has no effect on Tree Support

AngusX74, 2023-06-23:
### Application Version

5.4.0-beta.1

### Operating System

Windows 10

### Printer

Creality Ender 3 V2 Neo

### Reproduction steps

I want to slice a model that has overhanging structures and for that I want to use tree supports. Small overhangs with only a few lines should not be supported.

### Actual results

Minimum Support Area has no effect when using Tree supports.
Any area with an angle > Support Overhang Angle is supported, regardless of the value set. 

The problem occurs with every model.

### Expected results

Areas smaller than specified in Minimum Support Area should not be supported.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The problem occurs with every model where tree supports are used.

MariMakes, 2023-07-05:
Hey @AngusX74,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm having some trouble reproducing your issue. 
If I change the Minimum Support Area in my model I'm seeing that it's applied correctly.

Do you have a project file for us that shows your issue? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

AngusX74, 2023-07-06:
Hi @MariMakes,

I have attached an example project. In the example, the support should be printed only in the center of the sphere, but not under the narrow thread on the outer edge. But even with 10000 mm² Minimum Support Area, the support is still placed everywhere. With this setting, I would have expected that no support at all would be generated anymore, since no overhang is that large.
Or did I misunderstand something or set it wrong?
Just reproduced with the current version 5.4.

[Support_Problem.zip](https://github.com/Ultimaker/Cura/files/11970514/Support_Problem.zip)

ThomasRahm, 2023-07-09:
You are correct, this is not working as it should (at least by my understanding of what the settings should be doing).
1.  Support placed further down because of X/Y Overrides Z is never checked against minimum support area.
2. To support large areas when interface is disabled, i re-use the code that generates areas where roof is placed, but do not check against minimum support area, but minimum roof area (which is wrong). This should only be causing a similar problem for pretty large areas (larger than 100 mm^2).

In the exact settings given here 2. does not occur (but it may not be obvious that when roof can be generated the `minimum support roof area` has also to be increased, as otherwise roof will generate that will be supported by trees)

In short @AngusX74 If you set `Support Distance Priority` to `Z Overrides X/Y` and also increase the `Minimum Support Roof Area` it should behave as expected.

MariMakes, 2023-07-10:
Thanks for the update and the project file. 
This indeed illustrates your problem really well 😊 

Also thanks for he analysis @ThomasRahm 💕 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞 

I also found another workaround to get you unstuck. You can assign the Support Blocker properity to all kinds of meshes. If in your case I use a donut shaped mesh, I can just block the threads but leave space in the middle to support the roof. 
That would look like this:
[image]
[image]

You can download the project here: 
[Mari_CE3E3V2_Support_Problem.zip](https://github.com/Ultimaker/Cura/files/12000184/Mari_CE3E3V2_Support_Problem.zip)

AngusX74, 2023-07-11:
Hello,

thanks for your help.
Your solution works @MariMakes  (I didn't know that I can use any imported object as a support blocker). But for objects with more complex structures you would probably have to place a lot of support blockers anyway.

The solution of @ThomasRahm does not seem to work for me (see attachment Support_Problem_2.zip). I have made the recommended settings for `Distance Priority` (Z Overrides X/Y) and `Minimum Support Roof Area` (all minimum areas on 1000 mm²) and still support is generated at the edge of the hemisphere.

[Support_Problem_2.zip](https://github.com/Ultimaker/Cura/files/12018510/Support_Problem_2.zip)

ThomasRahm, 2023-07-13:
It works with 10000 mm² for `Minimum Support Roof Area` and Minimum Support Area. 1000 mm² are only 10 cm², and it seems to me that this area may be larger than this. If it is not please tell me and i will look into it more.

AngusX74, 2023-07-13:
Thank you for your help @ThomasRahm. You are right. I have calculated the ring area once and come to 1356 mm². So I was actually below the area size with my settings. However, no support is printed in the center of the sphere, because the area in the center is also smaller.
In the meantime, I'm not sure if I could achieve what I want with the `Minimum Support Area`. The goal was to print the small overhang of the ring without support, but since the complete ring area is considered the value would have to be chosen much too large. I think in this case the solution of @MariMakes is actually the best.
Thanks again to both of you for the tips.

ThomasRahm, 2023-07-15:
@AngusX74 Can you reopen this issue. The bug still exist for X/Y overrides Z.

AngusX74, 2023-07-15:
@ThomasRahm's workaround works, but he's right that the misbehavior with X/Y overrides Z still exists.

ThomasRahm, 2023-07-19:
@MariMakes
The commit https://github.com/ThomasRahm/CuraEngine/commit/810d6ea373f844f47b4b4990088899f250ea242a should fix the bugs in the engine. I just have not yet got around to update my development instance to 5.4, so i don't think i can currently easily cherry pick it and do a pull request for it.

I assume that the minimum support area is to be applied to the part of the overhang that is projected down (because of X/Y overrides Z), and not the whole overhang.

ToasterUwU, 2024-12-06:
Is this considered fixed? Because im on 5.9 and have this exact issue now. I also have a vaguely half globe shaped file where setting "Minimum Support Area" has no effect. The object I have fits in my palm and even with a value of 999999999999 mm² there is no effect.

I have it both with X/Y overrides Z and the other way around. 

If you want to test it, download the 3MF file from the following link. I have only tried the Roll-up case V2 USB-A file.
https://www.printables.com/model/697891-pixel-watch-2-3-charger-with-cable-winder-and-plug/files

ThomasRahm, 2024-12-07:
Please share it as a project file if possible, as then it contains your settings.
There are cases where what you are describing could be fine (e.g. if the support supports interface), but other cases where this would indicate that something is broken, so having the settings matters here imo.

ToasterUwU, 2024-12-07:
Sure thing @ThomasRahm, here the project file with the absurd big 99999999999 value already applied. The little rings of supports all around the base of this thing just shouldnt exist when using this setting as far as i understand it. In the description i read for what this setting does it specificially calls out tiny supports that dont do much and arent neccesary. In this specific case they do absolutely nothing and if anything are a risk of clumps building up and ruining the print early, so thats why i wanted to remove them.

[EN4PRO_Roll-up case V2 - USB-A.zip](https://github.com/user-attachments/files/18048841/EN4PRO_Roll-up.case.V2.-.USB-A.zip)

ThomasRahm, 2025-02-07:
@ToasterUwU  Sorry for the late response, have been busy for a bit.

When support roof is active, the `Minimum Support Interface Area` also needs to be set accordingly. If only the `Minimum Support Area` then roof will be placed which will then be supported. If only the `Minimum Support Interface Area` is increased, then there wont be any roof generated, meaning normal tree support will be placed.
If both are increased, it works as expected.

This is something that may not be as obvious as expected, as it comes up from time to time, so I think it should be added to the description of those settings.

ToasterUwU, 2025-02-07:
@ThomasRahm that makes sense. I agree, this should definetly be in the description