Ultimaker/Cura #15898: Slicing error

jeanluc-waurzyczka, 2023-06-24:
### Application Version

5.4.0-beta.1

### Operating System

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

I want to slice with the 5.4 beta 1. The result is "Slicing failed" I added the stl file. When I use the 5.3.1 it is working without any issue

### Actual results

Slicing failed

### Expected results

Generate Gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Arms-v2 - Gold.zip](https://github.com/Ultimaker/Cura/files/11857055/Arms-v2.-.Gold.zip)

jeanluc-waurzyczka, 2023-06-24:
Sorry it is the project file that I have added. Regards

MariMakes, 2023-07-17:
Hey @jeanluc-waurzyczka,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for the Project File 👍
I took a look at it in Cura 5.4 and I can reproduce your crash. 
I also see that if I reset the Minimum Wall Line Width, that I can slice your model succesfully, I hope that at least helps you to not be stuck anymore. 
[image]

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

jeanluc-waurzyczka, 2023-07-18:
Hello,

Thank you for your answer.

In my opinion there is no need for a quick fix as you can provide a bypass.

Another time thank you and take care.

Regards

 

 <https://user-images.githubusercontent.com/40423138/253987303-7ce05cbd-158e-40d4-960a-fe524f591e7a.png>

MariMakes, 2023-07-24:
Hey @jeanluc-waurzyczka,

Thanks for the update. 
Then I'll close this issue. 👍 

Good luck with your printjob!