Ultimaker/Cura #15899: Ejection Wiped micro SD! 😢

LeonWeaver, 2023-06-24:
### Application Version

5.3.0

### Operating System

MacOS Ventura 13.4

### Printer

Ender 3 V2

### Reproduction steps

To start I first realized there was a problem when I inserted the micro SD into the printer and it said I had "No Media" on the card which I knew must be the wrong, spoiler it was right the card was wiped! I'll just start from when I opened Cura and walk through every thing that happened. So I opened Cura, I added two stl. files then proceeded to slice them. Then I went to the printer got the micro SD brought it my MacBook and inserted it into a card reader that was a USB-A converted by a USB-A to USB-C because my Macbook only has USB-C. I then proceeded to export it and I accidentally export it TWICE, and then I ejected it and for some dumb reason I got curious and wondered what would happen if I ejected it TWICE, because it popped up TWO prompts to eject it and the first one was like normal but when I click the second one surprise surprise it threw a error. Also if it helps I inserted it into the printer while it was off then I turned it on and it then show a small prompt that media had fail to initiate... I've tried to replicate it without success! 😢

### Actual results

It completely wiped my micro SD but I'd have to reformat it or anything I was able to add new files like nothing had happened!?!

### Expected results

I would expect it will wipe your card as well when these steps are followed!? 🤷‍♂️ Maybe I'm just dumb? 😆🤷‍♂️ 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Gitbub Bug Report.3mf.zip](https://github.com/Ultimaker/Cura/files/11857067/Gitbub.Bug.Report.3mf.zip)
[GitHub Bug Report.zip](https://github.com/Ultimaker/Cura/files/11857078/GitHub.Bug.Report.zip)

MariMakes, 2023-07-25:
Hey @LeonWeaver,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This sounds like a super frustrating issue, but it's also the first time I've heard of it. 
I asked the developers to take a look but we couldn't figure out what was happening, if you pressed the Eject USB Safely correctly you should not have lost any of your files. 

I'm sorry you lost your files but we also don't know how to help with it. The only thing I can think of is making sure that if you load 3mfs/stls you load them from a stable place and if you want to save them you should give them a unique name every time. 

We believe we will not work on this anytime soon and will therefore defer this issue.
I'll close the issue, I hope you understand. If we get more reports on the issue we will probably take another look. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-07-25 by MariMakes

LeonWeaver, 2023-07-25:
Ya that is fine seemed like a one off type of thing and I probably just ejected incorrectly! Thanks for your time i appreciate it! 👍